leggedrobotics/viplanner #2: No module named 'omni.isaac.kit'

942411526, 2024-02-17:
Dear author,
thanks for your great work, there are some problems **ModuleNotFoundError: No module named 'omni.isaac.kit'**,
 we run the **/orbit.sh -p /home/yx/github_pr/viplanner/omniverse/standalone/viplanner_demo.py --scene warehouse --model_dir /home/yx/github_pr/viplanner/model** .  The **python -c "import omni.isaac"** is fine , but **python -c "import omni.isaac.kit"** error. Maybe the 
kit  lose?   our Isaac is isaac_sim-2022.2.1

942411526, 2024-02-17:
what version of Python do you use? **Package 'omni-isaac-contrib-tasks' requires a different Python: 3.7.4 not in '>=3.10'** . there is a problem **ModuleNotFoundError: No module named 'carb._carb'**, if we use the Python: 3.11

pascal-roth, 2024-02-17:
Hi,
This sounds more like a problem of your Isaac Orbit installation. The package that you are using is part of the Isaac Sim installation. Did you run `./orbit.sh -i -e` to install all the packages after creating the symlink for Isaac Sim and ViPlanner? The package has been tested with Python3.10 and Isaac Sim 2023.1.1

Please send the complete stack trace and all your system information if the error persists. Then, we can try to reproduce it.

pascal-roth, 2024-02-17:
Isaac Sim 2022.2.1 normally comes with an internal Python 3.7 version that the conda environment wraps around. I am not sure if it can result in conflicts if you use python 3.11 in your conda environment. I would recommend you to update to the latest Isaac Sim version (2023.1.1) which comes with Python 3.10 and keep the version. 

As a debug way, deactivate the conda environment. In this case, you install the additional packages directly in the Isaac Sim virtual env.

942411526, 2024-02-17:
> Isaac Sim 2022.2.1 normally comes with an internal Python 3.7 version that the conda environment wraps around. I am not sure if it can result in conflicts if you use python 3.11 in your conda [...]

thank you we will update the Isaac Sim and test again

942411526, 2024-02-17:
> /orbit.sh -p /home/yx/github_pr/viplanner/omniverse/standalone/viplanner_demo.py --scene warehouse --model_dir /home/yx/github_pr/viplanner/model

> thank you we will update the Isaac Sim and test again

I update the Isaac the problem was deal , but there is a new problem **2024-02-17 15:04:41 [137,985ms] [Error] [omni.isaac.orbit.utils.assets] Unable to perform Nucleus login on Omniverse. Assets root path is not set** when I run the **./orbit.sh -p /home/yx/github_pr/viplanner/omniverse/standalone/viplanner_demo.py --scene warehouse --model_dir /home/yx/github_pr/viplanner/model .** the widows will crash after wait a long time

pascal-roth, 2024-02-17:
You have to log in at the Omniverse launcher, ensure Nucleus is enabled, and the `localhost` drive is there. The warehouse env is referenced to it. Otherwise, you can also try the Carla environment, which is not referenced to Nucleus. If you downloaded it before today, please redownload it, as we uploaded an updated version.

942411526, 2024-02-18:
> updated version

updated version ? the code?

942411526, 2024-02-18:
lbar::play.SPACE, error code: Duplicated action definition
[14.072s] RTX ready
[14.399s] Simulation App Startup Complete
Traceback (most recent call last):
  File "/home/yx/github_pr/viplanner/omniverse/standalone/viplanner_demo.py", line 37, in <module>
    from omni.viplanner.config import (
  File "/home/yx/github_pr/viplanner/omniverse/extension/omni.viplanner/omni/viplanner/config/__init__.py", line 1, in <module>
    from .beat_the_planner_cfg import (
ModuleNotFoundError: No module named 'omni.viplanner.config.beat_the_planner_cfg'

pascal-roth, 2024-02-19:
Updated version of the carla mesh asset. 

Thanks for pointing out the issue, just remove the leftovers of the `beat_the_planner` configs.

pascal-roth, 2024-02-21:
Closing this issue now. Hope that everything is fixed. Otherwise please open a new issue.

closed 2024-02-21 by pascal-roth