leggedrobotics/viplanner #7: Dataset Request

BE-A-HERO, 2024-04-24:
Interesting work! May I ask will you release the dataset of the paper?  

This paper use several techniques in training set creation, it would be great if the author can also share the codes of dataset creation process. 

Thanks!!!

pascal-roth, 2024-04-24:
We are working on it, will be released before ICRA

jonyzhang2023, 2024-06-11:
> We are working on it, will be released before ICRA

Thanks for your work. And same question.
When will you share the codes of dataset creation process.

Xyunze, 2024-08-21:
Great work! I'm having trouble finding a way to construct the dataset during training. It would be greatly appreciated if you could share how to process the Matterport dataset.

PR #40 (2024-11-19): Adds data collection pipeline
# Description

Adds data collection pipeline. 

Fixes #7 #14 #36 

## Type of change

- New feature (non-breaking change which adds functionality)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./formatter.sh`
- [x] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
commit 7d481c1e (2024-11-19): Adds data collection pipeline (#40)

# Description

Adds data collection pipeline. 

Fixes #7 #14 #36 

## Type of change

- New feature (non-breaking change which adds functionality)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./formatter.sh`
- [x] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
closed 2024-11-19 by pascal-roth