leggedrobotics/viplanner #8: Carla texture file not found

YandongJi, 2024-04-30:
Hi,

When running demo with this script:
`./orbit.sh -p {VIPLANNER_DIR}/omniverse/standalone/viplanner_demo.py --scene carla --model_dir {MODEL_DIR}`

This error pops up: 
```2024-04-30 02:48:59 [48,443ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Town01_Opt/Looks/material_Game_Carla_Static_Wall_MI_WallMaster_MI_WallMaster/material_Game_Carla_Static_Wall_MI_WallMaster_MI_WallMaster' parameter 'diffuse_texture': References an asset that can not be found: '/home/pascal/viplanner/env/carla/town01/textures/MI_WallMaster_MI_WallMaster_D.bmp'```

Can I know where to adjust this path as well as the texture files? Thank you

pascal-roth, 2024-04-30:
Thanks for noticing, will look into it.

YandongJi, 2024-04-30:
Thanks for the great repo!

Actually when running the example code for Matterport, and Warehouse, they both have issues so far.

For Matterport, the error message is as follows:

```
Traceback (most recent call last):
  File "/MYPATH/viplanner/omniverse/standalone/viplanner_demo.py", line 135, in <module>
    main()
  File "/MYPATH/viplanner/omniverse/standalone/viplanner_demo.py", line 71, in main
    env = RLTaskEnv(env_cfg)
  File "/MYPATH/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/rl_task_env.py", line 86, in __init__
    super().__init__(cfg=cfg)
  File "/MYPATH/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/base_env.py", line 139, in __init__
    self.load_managers()
  File "/MYPATH/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/rl_task_env.py", line 129, in load_managers
    super().load_managers()
  File "/MYPATH/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/base_env.py", line 223, in load_managers
    self.observation_manager = ObservationManager(self.cfg.observations, self)
  File "/MYPATH/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/managers/observation_manager.py", line 41, in __init__
    super().__init__(cfg, env)
  File "/MYPATH/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/managers/manager_base.py", line 130, in __init__
    self._prepare_terms()
  File "/MYPATH/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/managers/observation_manager.py", line 254, in _prepare_terms
    obs_dims = tuple(term_cfg.func(self._env, **term_cfg.params).shape[1:])
  File "/MYPATH/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/mdp/observations.py", line 168, in height_scan
    return sensor.data.pos_w[:, 2].unsqueeze(1) - sensor.data.ray_hits_w[..., 2] - offset
  File "/MYPATH/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/ray_caster/ray_caster.py", line 104, in data
    self._update_outdated_buffers()
  File "/MYPATH/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/sensor_base.py", line 279, in _update_outdated_buffers
    self._update_buffers_impl(outdated_env_ids)
  File "/MYPATH/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/ray_caster/ray_caster.py", line 244, in _update_buffers_impl
    pos_w += self.drift[env_ids]
AttributeError: 'MatterportRayCaster' object has no attribute 'drift'
```
For warehouse, I hacked some bugs(removed int() at line 67 in observations.py), but issues are still popping up, the current message is:
`"*** TypeError: can only concatenate str (not "int") to str"` at line 79:
` mapping = torch.zeros((max(info[env_id].keys()) + 1, 3), dtype=torch.uint8, device=env.device)`

pascal-roth, 2024-04-30:
I am working on some updates right now, will have them ready in the next days. Will also check these issues, please make sure that you have the correct orbit version, then it should already

pascal-roth, 2024-05-06:
Tried the code again and it runs without an error, are you sure you checked out the correct orbit version as specified in the setup of the omniverse install script?

pascal-roth, 2024-05-15:
The Carla asset has been reuploaded, and the textures are now available. You can download the new asset [here](https://drive.google.com/file/d/1wZVKf2W0bSmP1Wm2w1XgftzSBx0UR1RK/view?usp=sharing).

Please let me know if there are still issues when running any of the examples.

YandongJi, 2024-05-22:
Thanks for the update! The assets are working now.

Actually, ply file cannot provide collision for Matterport, is there a good way to add collision mesh with current pipeline?

pascal-roth, 2024-05-22:
The ply file is just used to get the semantic data. You also load the mesh as the usd/obj file.

Normally collision meshes should be added to that mesh directly. Otherwise you can open the .usd file, select the matterport prim and in the properties select Add->Physics->Colliders. 

I hope this helps

pascal-roth, 2024-06-10:
Closing this issue for now, please reopen if there are any more questions

closed 2024-06-10 by pascal-roth