leggedrobotics/viplanner #10: ROS bulid

gftww, 2024-05-11:
Hi, There:
Considering the "catkin build viplanner_pkgs", can you explain the installation steps of "the viplanner repo should be linked into $HOME/catkin_ws/src" on the details, and it is located in manually installation of ROS Node?

Thank you very much

pascal-roth, 2024-05-12:
Hi, 
The structure assumes two directories, one for cloning the repo and the other for creating the Catkin workspace (that way, the same repo can be used for multiple workspaces, but that is not strictly necessary; you can also clone it directly in the workspace). In the case of two directories, link the cloned repo in the workspace's src directory. The command `catkin build viplanner_pkgs` then takes care of building all necessary packages at once.  

I hope that answers the first part of your question. For the second part " it is located in manually installation of ROS Node", could you explain in detail what you mean?

pascal-roth, 2024-06-10:
Closing this issue for now due to inactivity. Please reopen it, in the case there are any more issues

closed 2024-06-10 by pascal-roth