leggedrobotics/viplanner #13: errors in Dockerfile: detectron2 & mask2former

kozhukovv, 2024-05-23:
Hi there!
I'm very excited your work and would like to test it out at Orbit. I'm dev container for work, and when installing everything I have problems with the detectron2 and mask2former libraries. When I'm run this command:
```
./orbit.sh -p {VIPLANNER_DIR}/omniverse/standalone/viplanner_demo.py --scene warehouse --model_dir {MODEL_DIR}
```
i get this error:
```
[Warning] Pre-trained ResNet50 models cannot be used since detectron2 not found
[Warning] Pre-trained ResNet50 models cannot be used since mask2former not found
```
Unfortunately it seems that it is not possible to install library data inside a container. Could you tell me if you have any working cases of launching your model from docker to Orbit?

pascal-roth, 2024-05-23:
Hi,
Thanks a lot for using our work! These are actually no errors, the demo should run without problems. Mask2Former and Detectron are only required during inference on the hardware. The warnings also appear during the training, as there is some code structure that started to use RGB instead of semantics for which we wanted to use the semantic pre-trained backbone. 

I hope this answers your question.

kozhukovv, 2024-05-23:
Okay, now everything is a little clearer, thank you very much for the answer!

pascal-roth, 2024-05-24:
Great, let us know if there are any other questions

closed 2024-05-24 by pascal-roth