leggedrobotics/viplanner #14: Training Data Generation

AndyCao1125, 2024-06-06:
Dear authors,

Thank you for your work. I am experiencing the same issue as mentioned in #7 . Could you please let us know when the details of the training data generation will be released? Additionally, is there any plan to open-source the training data used in this work?

Thanks!

pascal-roth, 2024-06-10:
Hi, 

First of all, thanks for using our work. Our timeline got delayed due to the release of Isaac Sim 4.0. We hope to make the official release next week. 

The dataset itself will not be released, as the majority is from Matterport, which is protected by a License agreement (you can get access [here](https://niessner.github.io/Matterport/).

heyshouzi, 2024-09-04:
Since there is no published method for building viplanner formatted data, does that mean I can't use matterport for training so far.How about CARLA and warehouse?

closed 2024-11-19 by pascal-roth