leggedrobotics/viplanner #16: How to modify the simulation map

Ruihyw, 2024-06-18:
Hi!, thank you for the great work of your team！
I have reconstructed the cost map of Carla Town7 and trained the model.But i don't know how to test the model.
So,how can i modify the 'carla.config' to achieve the effect of town1 in the demo program 'viplanner_demo.py'
looking for ur reply,sincerely

Ruihyw, 2024-06-19:
> Hi, Thanks for using our work. Follow the steps:
> 
>     1. Export the town to USD (see the Omniverse unreal engine plugin).
> 
>     2. Once you have the USD, update the path in the config and adjust the start and goal points of the planner.
> 
>     3. Furthermore, for the town,  you have to map the meshes to semantic classes. This can be done by defining a YAML file similar to the one for town 01. The meshes are then traversed by their name, and if the name fits any of the defined string sections, the class is assigned.
> 
> 
> Once you have done these steps, you can replicate the demo. If there are any more questions, please let us know.

Thank you very much！i'll try it

Ruihyw, 2024-06-21:
[image]
sry to bother you again,i still have some question:
1.when I reconstruct the environment,i found the "cloud.ply' is less than 3MB size,looks like the pic above ,is this correct?
2.I changed the color rgb value of each type in the "viplanner_sem_meta.py" file to match my semantic data in "Tow7",but,when I run the "cost_builder.py",only less than 1%of the points match,and the generated cost map is strange.
[image]

Ruihyw, 2024-06-21:
[image]
this is the cost-map

Ruihyw, 2024-06-21:
I just changed the "VIPLANNER_SEM_META" and the "cls_order"
VIPLANNER_SEM_META = [
    # TRAVERSABLE SPACE ###
    # traversable intended
    {
        "name": "sidewalk",#check
        "loss": TRAVERSABLE_INTENDED_LOSS,
        "color": [244, 35, 232],
        "ground": True,
    },
    {
        "name": "crosswalk",#check
        "loss": TRAVERSABLE_INTENDED_LOSS,
        "color": [157, 234, 50],
        "ground": True,
    },
    {
        "name": "sand",#check
        "loss": TRAVERSABLE_INTENDED_LOSS,
        "color": [152, 251, 152],
        "ground": True,
    },
    {
        "name": "road",#check
        "loss": TRAVERSABLE_INTENDED_LOSS,
        "color": [128, 64, 128],
        "ground": True,
    },
    # traversable not intended

    {
        "name": "terrain",
        "color": [255, 255, 0],
        "loss": TERRAIN_LOSS,
        "ground": True,
    },

    # OBSTACLES ###
    # human
    {
        "name": "person",#check
        "color": [255, 0, 0],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    {
        "name": "anymal",#truck check
        "color": [0, 0, 70],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    # vehicle
    {
        "name": "vehicle",#check
        "color": [0, 0, 142],
        "loss": OBSTACLE_LOSS,  
        "ground": False,
    },
    {
        "name": "motorcycle",#check
        "color": [0, 0, 230],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    {
        "name": "bicycle",#check
        "color": [119, 11, 32],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    # construction
    {
        "name": "building",#check
        "loss": OBSTACLE_LOSS,
        "color": [70, 70, 70],
        "ground": False,
    },
    {
        "name": "bridge",  #barn check
        "loss": OBSTACLE_LOSS,
        "color": [170, 120, 50],
        "ground": False,
    },
    {
        "name": "fence",#check
        "color": [190, 153, 153],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    # object
    {
        "name": "pole",#check
        "color": [153, 153, 153],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    {
        "name": "traffic_sign",#check
        "color": [220, 220, 0],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    {
        "name": "traffic_light", #check
        "color": [250 , 170, 30],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    {
        "name": "bench",#ware dianxian check
        "color": [55, 90, 80],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    # nature
    {
        "name": "vegetation",#check
        "color": [107, 142, 35],
        "loss": TERRAIN_LOSS,
        "ground": False,
    },
    {
        "name": "water_surface",#zhongjian lanse de check
        "color": [45, 60, 150],
        "loss": OBSTACLE_LOSS,
        "ground": True,
    },
    # sky
    {
        "name": "sky",#check
        "color": [70, 130, 180],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    {
        "name": "background",#mountain check
        "color": [55, 90, 80],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    # void outdoor
    {
        "name": "dynamic",
        "color": [32, 0, 32],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
    {
        "name": "static",  # also everything unknown
        "color": [0, 0, 0],
        "loss": OBSTACLE_LOSS,
        "ground": False,
    },
]

cls_order = [
        ["sky", "background","dynamic", "static"],
        [
            "building",
            "fence",
            "water_surface",
            "bridge"
        ],  # 
        [
            "pole",
            "traffic_light",
            "traffic_sign",
            "bench"
        ],
        [ "terrain", "vegetation"],
        ["sidewalk", "crosswalk","sand","road"],
        ["person", "anymal", "vehicle", "motorcycle", "bicycle"],
    ]

pascal-roth, 2024-06-21:
Hi, no the reconstruction does not look correct. The mesh should be nicely recognizable. Do the rendered images look as expected, and do the poses make sense?  With the given environment reconstruction (basically just the lines), the strange costmap makes sense as it can only recognize these lines and fill the rest of the space. 

I would suggest double-checking how the images were created and that they are generated densely over the entire map so that the reconstruction results look like the mesh.

Ruihyw, 2024-06-23:
Thank you very much for your patient reply! I still have two questions
1.I placed a self-driving vehicle inside the Carla UE4 to get the dataset, which means that the vehicle will only drive along the road and take pictures, could this be the reason why the reconstruction fails?
Also.
2.I noticed that the reconstructed lines are actually blue sky and some green vegetation, does this mean that the reconstruction has something to do with the settings related to semantics, and if so what modifications do I need to make?

Ruihyw, 2024-06-23:
> Hi, no the reconstruction does not look correct. The mesh should be nicely recognizable. Do the rendered images look as expected, and do the poses make sense? With the given environment [...]

Thank you very much for your patient reply! I still have two questions
1.I placed a self-driving vehicle inside the Carla UE4 to get the dataset, which means that the vehicle will only drive along the road and take pictures, could this be the reason why the reconstruction fails?
Also.
2.I noticed that the reconstructed lines are actually blue sky and some green vegetation, does this mean that the reconstruction has something to do with the settings related to semantics, and if so what modifications do I need to make?

pascal-roth, 2024-06-23:
Did you adjust all the camera parameters for your recorded data? If they don't fit, then it is possible that your reconstruction does not give meaningful results. I would propose to export the assets to omniverse and then do the data collection there with the defined sensor parameters.

Ruihyw, 2024-06-23:
> Did you adjust all the camera parameters for your recorded data? If they don't fit, then it is possible that your reconstruction does not give meaningful results. I would propose to export the assets to omniverse and then do the data collection there with the defined sensor parameters.

yes，I get the internal and external parameters of the camera and store them in the correct paths，
[image]
[image]
[image]

pascal-roth, 2024-06-26:
I don't know how it fails then. I would propose to take two depth cameras close to each other and do the reconstruction to check if that looks correct.

Ruihyw, 2024-06-27:
> I don't know how it fails then. I would propose to take two depth cameras close to each other and do the reconstruction to check if that looks correct.

Hi！Thank you for your patience!I solved the problem of 3D reconstruction with only trajectories,I changed this line of code
          [image]

to:
            img_array = cv2.imread(img_path,cv2.IMREAD_ANYDEPTH)
And then the reconstruction results have content but are still wrong, could it be due to the scale factor?

[image]

And considering that the reconstruction results are still bad, could it be because the two neighboring images I used are only 0.02s apart, should I extend the sampling time
[image]
Looking forward to your reply！

pascal-roth, 2024-07-09:
Sorry for the late reply. 

Are you using the scale factor while recording the depth images? We introduced it when using png depth images in order to achieve a higher resolution. In the case you  didn't multiple the depth images during the recording with the scale factor, then you have to set it during the reconstruction to 1. 

That the images are 0.02 sec apart should not be issue. However, with more time it might be easier to see if the results are correct.

Ruihyw, 2024-07-22:
Hi!I have successfully generated the cost-map！
but when I run the training program I get the following error, 9929 is the number of depth maps I have captured
[image]
btw,What parts of the "learning-cfg" should I change, and also what should be the correct file structure for the training process?
Sincerely looking forward to your reply！

pascal-roth, 2024-07-25:
That's great to hear, could you share what you did to fix it? If there are any code fixes, please make a PR to the repo.

It is very hard to debug those issues from a distance. I suggest setting a debugger at that point. The error means that the number of depth images is different from the number of passed odometry points. Please double-check that these are the same. 

I cannot tell you what you must change to make it work. Generally, there should not be too many changes necessary. 

What do you mean by "correct file structure"?

Ruihyw, 2024-07-26:
> That's great to hear, could you share what you did to fix it? If there are any code fixes, please make a PR to the repo.
>
> It is very hard to debug those issues from a distance. I suggest setting a [...]

Hi！
1.Here is how I solved the problem
 I modified the way to save the depth map, I get the depth map from Carla and decode it and then save it as npy file, after that I set the scale factor to 1.In addition, since Carla uses a right-handed coordinate system and Open3d uses a left-handed coordinate system, I gave the y-axis the opposite number of external parameters to ensure that the 3D reconstruction works properly. I don't see this as a problem with the code, but rather part of the manipulation I need to do during data acquisition.
2.I'm confused as to why there is a mismatch, and since I have 9930 images with serial numbers 0-9929, the mismatch that occurs here is because one dimension is 8829 and the other is 8830.This problem was also solved after a modification
3.Since I've only rebuilt the Town01 environment so far, I'd like to utilize only that environment for training. So I deleted 'env_list_combi' from 'train.py' and default_factory from 'learning_cfg.py' other All elements, only Town01 is kept, I'm not sure if this is correct or not
[image]
[image]
4.I mean how should I set the file paths, I have very strange paths now, should I put all the environments in the data directory and then change the file_path in learning_cfg.py to the path of data?
5.One last question, I'm training the model right now but the loss function is changing strangely (both train_loss and val_loss)and I'm not sure if this is correct or not.The training loss stops decreasing after the initial few epochs are reduced to about 5
[image]

Ruihyw, 2024-07-26:
[image]
This is my decoding operation for the depth image, after which I just set the depth factor to 1

Ruihyw, 2024-07-26:
[image]

pascal-roth, 2024-07-26:
to the different points:
1. nice, yes the different conventions for the rotations can lead to errors
2. modification to your data acquisition or the viplanner code?
3. the changes made should work, basically you only have to delete the other environments
4. the file paths to the data and the models can be set with the `file_path` parameter. Then your paths should be okay
5. I have not trained a policy just on a single environment, so I am a bit unsure. Generally, all the loss terms to decrease, which is fine. What happens when you apply it in the town?

Ruihyw, 2024-07-31:
> to the different points:
>
> 1. nice, yes the different conventions for the rotations can lead to errors
> 2. modification to your data acquisition or the viplanner code?
> 3. the changes made should work, [...]

1.Thank you very much for your patient reply！ 
2.I have modified both:
first for data collection, I linearly decoded the raw depth information from Carla and stored it in npy files. 
Secondly, I modified the depth factor in 'depth_reconstruction.py' to 1 and added part of the code for the coordinate system conversion
3.It is true that the training loss has been getting smaller, which is a good thing, but what is the reason that only the **goal loss**, the **height loss**, and the **motion loss** have actually been getting smaller, and the **obstacle loss has remained at a higher value**? Is this normal?
[image]
[image]

Ruihyw, 2024-07-31:
Also, when I test the program using the demo, the robot can **go right through objects like cars or walls**, at the same time I noticed that the objects in the environment are loaded in as a **'terrain'** as a whole, is this the reason why each individual object has no collision volume?

closed 2024-08-26 by Ruihyw

fan-ziqi, 2024-10-18:
@Ruihyw May I ask how you prepared the reconstruction data?