leggedrobotics/viplanner #17: The comparison between iplanner and Viplanner

rookie0109, 2024-07-07:
This is excellent work, but I noticed that the trajectory of iplanner in the video shakes left and right, whereas Viplanner does not exhibit this phenomenon. Could you please explain the main improvement strategies Viplanner has implemented to address this issue? Is it related to the construction of the cost map?

[image]

pascal-roth, 2024-07-09:
Hi, 

Thanks a lot for using our work. The reason for the shaking behavior is related to the cost-map and the absence of memory. 

Basically both planner replan in every time step. However, while for ViPlanner the cost-surface is more restricted, iPlanner doesn't perceive a difference between the terrain and has a large open low-cost space where the planner can predict a variety of paths that reach the goal with an overall low cost. Being more restricted for ViPlanner is fluctuation is less which leads to less jumps. 

I hope this explanation is sufficient. In the case there are any other questions, please don't hesitate to contact us.

rookie0109, 2024-07-12:
Thanks for your reply!

closed 2024-07-12 by rookie0109