leggedrobotics/viplanner #18: How to achieve depth reconstruction for matterport3D?

CrystalSixone, 2024-07-08:


pascal-roth, 2024-07-09:
Hi,

First of all thanks for using our work. To answer your points:

1. We are working on https://github.com/leggedrobotics/isaac-nav-suite (not public yet, currently updated to latest Isaac 4.0 version) which will automate the data generation and which will also have an example code for the 3D reconstruction. 

2. This ply file is delivered with the rest of the matterport files. It should be under the directory "house_segmentation" if you download the zip file for an environment. 

I hope that helps you to get started. If there are any more issues, please let us know.

CrystalSixone, 2024-07-11:
Thank you very much! Looking forward to your new work 👍

closed 2024-07-11 by CrystalSixone