leggedrobotics/viplanner #19: Run roslaunch viplanner_node viplanner.launch in the docker environment, the m2f_inference.py meet error

tenthousandrain, 2024-07-09:
The error messages:
NODES
  /
    pathFollower (path_follower/path_follower_node)
    viplanner_node (viplanner_node/viplanner_node.py)

auto-starting new master
process[master]: started with pid [193]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9869e4a6-3dcb-11ef-a3cb-48b02dea7a8a
process[rosout-1]: started with pid [206]
started core service [/rosout]
process[pathFollower-2]: started with pid [213]
process[viplanner_node-3]: started with pid [214]
/root/.local/lib/python3.8/site-packages/torchvision/io/image.py:13: UserWarning: Failed to load image Python extension: 
  warn(f"Failed to load image Python extension: {e}")
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (2.2.2) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
[Warning] Pre-trained ResNet50 models cannot be used since detectron2 not found
[Warning] Pre-trained ResNet50 models cannot be used since mask2former not found
[INFO] [1720513017.604229]: Get param viplanner_node/main_freq
[INFO] [1720513017.607299]: Get param viplanner_node/image_flip
[INFO] [1720513017.610182]: Get param viplanner_node/conv_dist
[INFO] [1720513017.613056]: Get param viplanner_node/max_depth
[INFO] [1720513017.615842]: Get param viplanner_node/overlap_ratio_thres
[INFO] [1720513017.618825]: Get param viplanner_node/depth_zero_ratio_thres
[INFO] [1720513017.621779]: Get param viplanner_node/model_save
[INFO] [1720513017.624566]: Get param viplanner_node/m2f_cfg_file
[INFO] [1720513017.627485]: Get param viplanner_node/m2f_model_path
[INFO] [1720513017.630301]: Get param viplanner_node/depth_topic
[INFO] [1720513017.633150]: Get param viplanner_node/depth_info_topic
[INFO] [1720513017.635897]: Get param viplanner_node/rgb_topic
[INFO] [1720513017.638585]: Get param viplanner_node/rgb_info_topic
[INFO] [1720513017.657058]: Get param viplanner_node/goal_topic
[INFO] [1720513017.660632]: Get param viplanner_node/path_topic
[INFO] [1720513017.666303]: Get param viplanner_node/m2f_timer_topic
[INFO] [1720513017.673315]: Get param viplanner_node/depth_uint_type
[INFO] [1720513017.677351]: Get param viplanner_node/compressed
[INFO] [1720513017.687420]: Get param viplanner_node/mount_cam_frame
[INFO] [1720513017.692036]: Get param viplanner_node/robot_id
[INFO] [1720513017.715265]: Get param viplanner_node/world_id
[INFO] [1720513017.719250]: Get param viplanner_node/is_fear_act
[INFO] [1720513017.722186]: Get param viplanner_node/buffer_size
[INFO] [1720513017.725277]: Get param viplanner_node/angular_thread
[INFO] [1720513017.728445]: Get param viplanner_node/track_dist
[INFO] [1720513017.731409]: Get param viplanner_node/joyGoal_scale
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mmengine/config/config.py", line 109, in __getattr__
    value = super().__getattr__(name)
  File "/usr/local/lib/python3.8/dist-packages/addict/addict.py", line 67, in __getattr__
    return self.__getitem__(item)
  File "/usr/local/lib/python3.8/dist-packages/mmengine/config/config.py", line 138, in __getitem__
    return self.build_lazy(super().__getitem__(key))
  File "/usr/local/lib/python3.8/dist-packages/mmengine/config/config.py", line 105, in __missing__
    raise KeyError(name)
KeyError: 'layer_cfg'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/catkin_ws/devel/lib/viplanner_node/viplanner_node.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/root/catkin_ws/src/viplanner/ros/planner/src/viplanner_node.py", line 1035, in <module>
    node = VIPlannerNode(args)
  File "/root/catkin_ws/src/viplanner/ros/planner/src/viplanner_node.py", line 65, in __init__
    self.m2f_inference = Mask2FormerInference(
  File "/root/catkin_ws/src/viplanner/ros/planner/src/m2f_inference.py", line 32, in __init__
    self.model = init_detector(config_file, checkpoint_file, device="cuda:0")
  File "/root/.local/lib/python3.8/site-packages/mmdet/apis/inference.py", line 66, in init_detector
    model = MODELS.build(config.model)
  File "/usr/local/lib/python3.8/dist-packages/mmengine/registry/registry.py", line 570, in build
    return self.build_func(cfg, *args, **kwargs, registry=self)
  File "/usr/local/lib/python3.8/dist-packages/mmengine/registry/build_functions.py", line 232, in build_model_from_cfg
    return build_from_cfg(cfg, registry, default_args)
  File "/usr/local/lib/python3.8/dist-packages/mmengine/registry/build_functions.py", line 121, in build_from_cfg
    obj = obj_cls(**args)  # type: ignore
  File "/root/.local/lib/python3.8/site-packages/mmdet/models/detectors/mask2former.py", line 22, in __init__
    super().__init__(
  File "/root/.local/lib/python3.8/site-packages/mmdet/models/detectors/maskformer.py", line 36, in __init__
    self.panoptic_head = MODELS.build(panoptic_head_)
  File "/usr/local/lib/python3.8/dist-packages/mmengine/registry/registry.py", line 570, in build
    return self.build_func(cfg, *args, **kwargs, registry=self)
  File "/usr/local/lib/python3.8/dist-packages/mmengine/registry/build_functions.py", line 232, in build_model_from_cfg
    return build_from_cfg(cfg, registry, default_args)
  File "/usr/local/lib/python3.8/dist-packages/mmengine/registry/build_functions.py", line 121, in build_from_cfg
    obj = obj_cls(**args)  # type: ignore
  File "/root/.local/lib/python3.8/site-packages/mmdet/models/dense_heads/mask2former_head.py", line 104, in __init__
    self.num_heads = transformer_decoder.layer_cfg.cross_attn_cfg.num_heads
  File "/usr/local/lib/python3.8/dist-packages/mmengine/config/config.py", line 113, in __getattr__
    raise AttributeError(f"'{self.__class__.__name__}' object has no "
AttributeError: 'ConfigDict' object has no attribute 'layer_cfg'
[viplanner_node-3] process has died [pid 214, exit code 1, cmd /root/catkin_ws/devel/lib/viplanner_node/viplanner_node.py __name:=viplanner_node __log:=/root/.ros/log/9869e4a6-3dcb-11ef-a3cb-48b02dea7a8a/viplanner_node-3.log].
log file: /root/.ros/log/9869e4a6-3dcb-11ef-a3cb-48b02dea7a8a/viplanner_node-3*.log
^C[pathFollower-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
root@ubuntu:~/catkin_ws/devel#

pascal-roth, 2024-07-09:
Hi, 

Thanks for using our work. This looks a lot like a version issue for me. Can you tell me which version of mmengine are you using?

tenthousandrain, 2024-07-10:
Thanks for your reply!
This is the version of mmengine in the docker:
mmcv                          2.0.0
mmdet                         3.3.0
mmengine                      0.10.4

tenthousandrain, 2024-07-22:
> Hi,
> 
> Thanks for using our work. This looks a lot like a version issue for me. Can you tell me which version of mmengine are you using?

I'm sorry, but is there any solution to slove this error?

pascal-roth, 2024-07-25:
The mmcv version seems correct. At the time of the project, the corresponding versions were:
mmdet 3.1.0
mmengine 0.8.4

Please check if these changes fix your issues.

pascal-roth, 2024-09-04:
I assume the issue has been fixed. Closing it now due to inactivity. Please reopen in the case the issue still persists

closed 2024-09-04 by pascal-roth