leggedrobotics/viplanner #20: env_nums > 1

hutslib, 2024-07-24:
is it possible to set env_nums > 1to use multiple environment to training?

pascal-roth, 2024-07-25:
Hi, 

I assume you mean the `num_envs` passed to the `TerrainSceneCfg`. This is just done for the demo scripts. The training itself is not performed online, s.t. there is no notion of multiple environments for training. It uses offline collected data and  runs batched, with a batch size depending on the size of your GPU.

Please let us know if that answers your question.

hutslib, 2024-07-25:
Hi, clear explanation of the pipeline.
One more question.
Also, when the data is collected, is num_envs set as 1? The question is generated based on my trial of using isaaclab.
I load your environment  (e.g. a Carla town), and set num_envs > 1.  The problem is the image collected by the robot can see robots in other environments.

pascal-roth, 2024-07-25:
For the data collection, I am using a pipeline that just spawns a camera at different points in the space without actually using a robot. Consequently, many can be rendered at the same time without any robots in the image.

hutslib, 2024-07-25:
Thanks for your clear explanation.

pascal-roth, 2024-07-25:
No worries, let me know if you have any further questions. 

Will close this issue for now.

closed 2024-07-25 by pascal-roth