leggedrobotics/viplanner #26: The adaptability of the model to camera perspectives

yzysmile, 2024-10-09:
Thanks for your the outstanding work!

The original paper mentioned that "The corresponding images at each sampled point are then rendered at angles that maximize the coverage of the traversable space." 
Does it mean capturing images in a virtual environment from different camera perspectives?

yzysmile, 2024-10-09:
> Thanks for your the outstanding work!
> 
> The original paper mentioned that "The corresponding images at each sampled point are then rendered at angles that maximize the coverage of the traversable space." Does it mean capturing images in a virtual environment from different camera perspectives?

To be more specifically, for example, the camera's perspectives at current sample point may different from the last sample point.Is it true?

pascal-roth, 2024-10-12:
> Thanks for your the outstanding work!
> 
> The original paper mentioned that "The corresponding images at each sampled point are then rendered at angles that maximize the coverage of the traversable space." Does it mean capturing images in a virtual environment from different camera perspectives?

Thanks a lot for using our work! 

During the sampling, we check the average distance of the depth images. To maximize the coverage of the traversable space, images are filtered where the average distance is below a certain threshold. Generally, the camera orientation will differ for each sample point, as it is sampled randomly in certain ranges. Does that answer your question.

yzysmile, 2024-10-14:
> Thanks a lot for using our work!
>
> During the sampling, we check the average distance of the depth images. To maximize the coverage of the traversable space, images are filtered where the average [...]

Will the camera look downing during sampling as our robot has a camera looking down at an angle about 10 degree? 
As the below picture shows, red coordinate represents a camera looking down at an angle about 10 degree.
[image]

pascal-roth, 2024-10-14:
We also tested it with a camera looking down up to 20 degrees and added certain randomization on top of that, there seemed to be no issue

pascal-roth, 2024-10-21:
@yzysmile did that work for you?

pascal-roth, 2024-10-22:
closing this issue now as it seems resolved

closed 2024-10-22 by pascal-roth

yzysmile, 2024-11-02:
> @yzysmile did that work for you?

yes, thanks for your reply. ☺