leggedrobotics/viplanner #27: ValueError: invalid literal for int() with base 10: '(0, 0, 0, 0)'

fan-ziqi, 2024-10-16:
When running `./orbit.sh -p {VIPLANNER_DIR}/omniverse/standalone/viplanner_demo.py --scene carla --model_dir {MODEL_DIR}`

```
Traceback (most recent call last):
  File "/home/ubuntu/workspaces/viplanner_ws/viplanner/omniverse/standalone/viplanner_demo.py", line 155, in <module>
    main()
  File "/home/ubuntu/workspaces/viplanner_ws/viplanner/omniverse/standalone/viplanner_demo.py", line 73, in main
    env = ManagerBasedRLEnv(env_cfg)
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 75, in __init__
    super().__init__(cfg=cfg)
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 136, in __init__
    self.load_managers()
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 115, in load_managers
    super().load_managers()
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 213, in load_managers
    self.observation_manager = ObservationManager(self.cfg.observations, self)
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/observation_manager.py", line 69, in __init__
    super().__init__(cfg, env)
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/manager_base.py", line 130, in __init__
    self._prepare_terms()
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/observation_manager.py", line 337, in _prepare_terms
    obs_dims = tuple(term_cfg.func(self._env, **term_cfg.params).shape)
  File "/home/ubuntu/workspaces/viplanner_ws/viplanner/omniverse/extension/omni.viplanner/omni/viplanner/viplanner/mdp/observations.py", line 65, in isaac_camera_data
    info = [
  File "/home/ubuntu/workspaces/viplanner_ws/viplanner/omniverse/extension/omni.viplanner/omni/viplanner/viplanner/mdp/observations.py", line 66, in <listcomp>
    {
  File "/home/ubuntu/workspaces/viplanner_ws/viplanner/omniverse/extension/omni.viplanner/omni/viplanner/viplanner/mdp/observations.py", line 67, in <dictcomp>
    int(k): VIPLANNER_SEM_META.class_color["static"]
ValueError: invalid literal for int() with base 10: '(0, 0, 0, 0)'
```

pascal-roth, 2024-10-16:
Hi @fan-ziqi,

thanks a lot for using our work. 

You are using ViPlanner with IsaacLab. However, it was developed which a much earlier version (still orbit and even earlier). Please checkout the commit:  477cd6b3f (described [here](https://github.com/leggedrobotics/viplanner/tree/main/omniverse#installation) and then try it again

fan-ziqi, 2024-10-16:
I have already ported all the APIs to IsaacLab. and I want to know what caused this issue.

Thanks~

pascal-roth, 2024-10-16:
Amazing that you ported the API. It would be great if you could make a PR with that so more people can profit from it. 

AFAIK, the API of the semantic observations changes at some point in IsaacSim, which might have affected the return values to a different type.  If you make the PR I can look into it but I am not sure about the exact changes now.

fan-ziqi, 2024-10-16:
This requires a lot of work, and after completing the document, I will initiate a PR

fan-ziqi, 2024-10-16:
PR has been opened #28

fan-ziqi, 2024-10-16:
There is, in addition, the issue mentioned in #8 still occurs in IsaacLab

```
...

2024-10-16 12:26:37 [20,823ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Looks/material_Game_Carla_Static_Static_Materials_DCarlacola_M_CarlaCola_M_CarlaCola/material_Game_Carla_Static_Static_Materials_DCarlacola_M_CarlaCola_M_CarlaCola' parameter 'diffuse_texture': References an asset that can not be found: './textures/M_CarlaCola_M_CarlaCola_D.bmp'
2024-10-16 12:26:37 [20,824ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Looks/material_Game_Carla_Static_Vehicles_GeneralMaterials_BrushedAluminiumWhite_Inst_BrushedAluminiumWhite_Inst/material_Game_Carla_Static_Vehicles_GeneralMaterials_BrushedAluminiumWhite_Inst_BrushedAluminiumWhite_Inst' parameter 'normalmap_texture': References an asset that can not be found: './textures/BrushedAluminiumWhite_Inst_BrushedAluminiumWhite_Inst_N.bmp'
2024-10-16 12:26:37 [20,824ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Looks/material_Game_Carla_Static_Vehicles_GeneralMaterials_BrushedAluminiumWhite_Inst_BrushedAluminiumWhite_Inst/material_Game_Carla_Static_Vehicles_GeneralMaterials_BrushedAluminiumWhite_Inst_BrushedAluminiumWhite_Inst' parameter 'diffuse_texture': References an asset that can not be found: './textures/BrushedAluminiumWhite_Inst_BrushedAluminiumWhite_Inst_D.bmp'
2024-10-16 12:26:37 [20,825ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Looks/material_Game_Carla_Static_Wall_Wall05_MI_Wall05_MI_Wall05/material_Game_Carla_Static_Wall_Wall05_MI_Wall05_MI_Wall05' parameter 'normalmap_texture': References an asset that can not be found: './textures/MI_Wall05_MI_Wall05_N.bmp'
2024-10-16 12:26:37 [20,825ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Looks/material_Game_Carla_Static_Wall_Wall05_MI_Wall05_MI_Wall05/material_Game_Carla_Static_Wall_Wall05_MI_Wall05_MI_Wall05' parameter 'diffuse_texture': References an asset that can not be found: './textures/MI_Wall05_MI_Wall05_D.bmp'
2024-10-16 12:26:37 [20,826ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Looks/material_Game_Carla_Static_Building_Materials_House01_M_House01WinFrames_M_House01WinFrames/material_Game_Carla_Static_Building_Materials_House01_M_House01WinFrames_M_House01WinFrames' parameter 'normalmap_texture': References an asset that can not be found: './textures/M_House01WinFrames_M_House01WinFrames_N.bmp'
2024-10-16 12:26:37 [20,826ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Looks/material_Game_Carla_Static_Building_Materials_House01_M_House01WinFrames_M_House01WinFrames/material_Game_Carla_Static_Building_Materials_House01_M_House01WinFrames_M_House01WinFrames' parameter 'diffuse_texture': References an asset that can not be found: './textures/M_House01WinFrames_M_House01WinFrames_D.bmp'
2024-10-16 12:26:37 [20,827ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Looks/material_Game_Carla_Static_Wall_Wall04_MI_Wall04_MI_Wall04/material_Game_Carla_Static_Wall_Wall04_MI_Wall04_MI_Wall04' parameter 'normalmap_texture': References an asset that can not be found: './textures/MI_Wall04_MI_Wall04_N.bmp'
2024-10-16 12:26:37 [20,827ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Looks/material_Game_Carla_Static_Wall_Wall04_MI_Wall04_MI_Wall04/material_Game_Carla_Static_Wall_Wall04_MI_Wall04_MI_Wall04' parameter 'diffuse_texture': References an asset that can not be found: './textures/MI_Wall04_MI_Wall04_D.bmp'
2024-10-16 12:26:37 [20,828ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Looks/material_Game_Carla_Static_Wall_Wall01_MI_Wall01_MI_Wall01/material_Game_Carla_Static_Wall_Wall01_MI_Wall01_MI_Wall01' parameter 'normalmap_texture': References an asset that can not be found: './textures/MI_Wall01_MI_Wall01_N.bmp'
2024-10-16 12:26:37 [20,828ms] [Error] [omni.hydra] [UsdToMdl] Prim '/World/Carla/terrain/Looks/material_Game_Carla_Static_Wall_Wall01_MI_Wall01_MI_Wall01/material_Game_Carla_Static_Wall_Wall01_MI_Wall01_MI_Wall01' parameter 'diffuse_texture': References an asset that can not be found: './textures/MI_Wall01_MI_Wall01_D.bmp'

...
```

fan-ziqi, 2024-10-17:
I‘ve found the changelog in `0.13.0 (2024-03-12)`

> Fixed the return shape of colored images for segmentation data types inside the [omni.isaac.lab.sensors.Camera](https://docs.robotsfan.com/isaaclab/source/api/lab/omni.isaac.lab.sensors.html#omni.isaac.lab.sensors.Camera) class. Earlier, the images were always returned as `int32`. Now, they are casted to `uint8` 4-channel array before returning if colorization is enabled for the annotation type.

It should be this change that caused the problem

fan-ziqi, 2024-10-17:
After setting colorize_semantic_segmentation to False in semantic_camera cfg, this leads to a new bug

```
Traceback (most recent call last):
  File "/home/ubuntu/workspaces/viplanner_ws/viplanner/omniverse/standalone/viplanner_demo.py", line 153, in <module>
    main()
  File "/home/ubuntu/workspaces/viplanner_ws/viplanner/omniverse/standalone/viplanner_demo.py", line 73, in main
    env = ManagerBasedRLEnv(env_cfg)
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 75, in __init__
    super().__init__(cfg=cfg)
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 136, in __init__
    self.load_managers()
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 115, in load_managers
    super().load_managers()
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 213, in load_managers
    self.observation_manager = ObservationManager(self.cfg.observations, self)
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/observation_manager.py", line 69, in __init__
    super().__init__(cfg, env)
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/manager_base.py", line 130, in __init__
    self._prepare_terms()
  File "/home/ubuntu/workspaces/viplanner_ws/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/observation_manager.py", line 337, in _prepare_terms
    obs_dims = tuple(term_cfg.func(self._env, **term_cfg.params).shape)
  File "/home/ubuntu/workspaces/viplanner_ws/viplanner/omniverse/extension/omni.viplanner/omni/viplanner/viplanner/mdp/observations.py", line 92, in isaac_camera_data
    return output.permute(0, 3, 1, 2)
RuntimeError: permute(sparse_coo): number of dimensions in the tensor input does not match the length of the desired ordering of dimensions i.e. input.dim() = 5 is not equal to len(dims) = 4
```

closed 2024-10-17 by fan-ziqi