leggedrobotics/viplanner #29: Installation with Docker image issues

SnoopyGeeek, 2024-10-17:
Thank you for your outstanding work. I am new to Robot Plannning and I have some issues in the installation step. I tired to run viplanner on the docker container of Jetson AGX Orin according to https://github.com/leggedrobotics/viplanner/blob/main/ros/README.md

However, I met some problems,
1) Inside the container, pytorch check cuda available return me False while outside the container return me True. This means that GPU is not available. 
[image]
2) Inside the container, how do I implement the step 4:
[image]
I tried to run it and it returned:
[image]
Could you plz fill me in?

pascal-roth, 2024-10-17:
Hi @SnoopyGeeek, 

Thanks a lot for using our work. 

To 1): a known reason that I remember is that the Nvidia runtime is not set correctly. Did you follow the steps described [here](https://github.com/leggedrobotics/viplanner/blob/main/ros/INSTALL.md#nvidia-jetson-docker). Otherwise, this is a very setup-dependent issue that is hard to solve remotely.

To 2): we did not make a catkin workspace under "/" but instead inside home. This basically means you create two directories outside of your docker under your home. These directories are "git" and "catkin_ws" whereas the repo should be cloned inside the "git" repo and then symlinked into the "catkin_ws/src" directory. Then you have the automatic binding into the docker container [here](https://github.com/leggedrobotics/viplanner/blob/61016e892366fdebc46640a4784b6c69f7e8a360/bin/run.sh#L23). You would then build the planner inside the container by running:

```bash
cd /home/catkin_ws
catkin build viplanner_pkgs
```

pascal-roth, 2024-10-21:
@SnoopyGeeek is your issue fixed with these changes?

pascal-roth, 2024-11-19:
Closing this issue now, assumed as solved. Please reopen in case the issue persists.

closed 2024-11-19 by pascal-roth