leggedrobotics/viplanner #31: anymal-c policy.pt not exist

heyshouzi, 2024-10-18:
[image]
FileNotFoundError: Policy file 'http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2/Isaac/Samples/Orbit/Policies/ANYmal-C/policy.pt' does not exist.

I am using isaac sim version 4.2  and I have downloaded the asset to the local path.

If the policy.pt file is not available over the Internet, where does it reside in the isaac sim asset?

fan-ziqi, 2024-10-18:
Please check #28

heyshouzi, 2024-10-18:
I  found the file locally under the path you mentioned above, but there was a problem similar to #28
``
`Traceback (most recent call last):
  File "/home/zhineng/wl/viplanner/omniverse/standalone/viplanner_demo.py", line 152, in <module>
    main()
  File "/home/zhineng/wl/viplanner/omniverse/standalone/viplanner_demo.py", line 122, in main
    obs = env.step(action=paths.view(paths.shape[0], -1))[0]
  File "/home/zhineng/wl/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/rl_task_env.py", line 199, in step
    self.obs_buf = self.observation_manager.compute()
  File "/home/zhineng/wl/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/managers/observation_manager.py", line 124, in compute
    obs_buffer[group_name] = self.compute_group(group_name)
  File "/home/zhineng/wl/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/managers/observation_manager.py", line 170, in compute_group
    obs: torch.Tensor = term_cfg.func(self._env, **term_cfg.params)
  File "/home/zhineng/wl/viplanner/omniverse/extension/omni.viplanner/omni/viplanner/viplanner/mdp/observations.py", line 93, in cam_position
    sensor: CameraData = env.scene.sensors[sensor_cfg.name].data
  File "/home/zhineng/wl/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/camera/camera.py", line 147, in data
    self._update_outdated_buffers()
  File "/home/zhineng/wl/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/sensor_base.py", line 275, in _update_outdated_buffers
    outdated_env_ids = self._is_outdated.nonzero().squeeze(-1)
RuntimeError: CUDA error: device-side assert triggered
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

2024-10-18 05:43:02 [326,046ms] [Error] [omni.physx.fabric.plugin] CUDA error: device-side assert triggered: ../../../extensions/runtime/source/omni.physx.fabric/plugins/DirectGpuHelper.cpp: 328
2024-10-18 05:43:02 [326,046ms] [Error] [omni.physx.fabric.plugin] CUDA error: device-side assert triggered: ../../../extensions/runtime/source/omni.physx.fabric/plugins/DirectGpuHelper.cpp: 331
2024-10-18 05:43:02 [326,046ms] [Error] [omni.physx.fabric.plugin] CUDA error: device-side assert triggered: ../../../extensions/runtime/source/omni.physx.fabric/plugins/DirectGpuHelper.cpp: 334
2024-10-18 05:43:02 [326,046ms] [Error] [omni.physx.fabric.plugin] CUDA error: device-side assert triggered: ../../../extensions/runtime/source/omni.physx.fabric/plugins/DirectGpuHelper.cpp: 337
2024-10-18 05:43:02 [326,046ms] [Error] [omni.physx.fabric.plugin] CUDA error: device-side assert triggered: ../../../extensions/runtime/source/omni.physx.fabric/plugins/DirectGpuHelper.cpp: 340
`

heyshouzi, 2024-10-18:
> I found the file locally under the path you mentioned above, but there was a problem similar to #28 `` `Traceback (most recent call last): File [...]

I don't know what happen.
open isaac sim,then click the play button, anymal stays there I wait for a while, and isaac sim flashes out.
I just updated isaac sim form version 2023.hotfix.1.1 to 4.2
`TORCH_USE_CUDA_DSA` to enable device-side assertions.`

pascal-roth, 2024-10-21:
Hi, 

thanks a lot for using our work. There has been a change in the return of the semantic camera when updating Isaac Sim. I just finished a fix and will push it together with the PR #28

heyshouzi, 2024-10-21:
Thanks for your great work,I   I should not need to manually configure anymal-c policy.pat after you  finish a fix?

pascal-roth, 2024-10-21:
No, after the merge the policy should be loaded automatically without the need to do anything from your side.

closed 2024-10-21 by heyshouzi