leggedrobotics/viplanner #36: [Question] semantic color space

fan-ziqi, 2024-10-29:
### Question

Every time I load the Carla environment, the color of the semantic map is different. How can I convert the semantic information into the color space of VIPlanner?

pascal-roth, 2024-10-29:
Hi Zigi, 

When do you observe the different colors? Typically, IsaacSim can assign random colors, but the class names are kept constant. The camera code of ViPlanner then takes the class names and converts them into colors, so that should be independent of what semantics you would see in the GUI.

Does that clarify the question?

fan-ziqi, 2024-10-30:
Thank you for your guidance. 

I created a new `CalraCamera(Camera)` class and mapped the colors as inputs for Carla's camera. The code has been uploaded [here](https://github.com/fan-ziqi/viplanner_data_generator/blob/4793244e400ba9014e3d758189073f32d084f105/extensions/omni.viplanner.importer/omni/viplanner/importer/sensors/carla_camera.py#L46-L70), and I hope it will be helpful for this project.

However, the costmap I generated is quite different from the one in the paper. Does the number of images used in reconstruction affect this? How should I set this value and the number of viewpoints?

[image]

pascal-roth, 2024-11-19:
Issue addressed as part of the new data generation code release

closed 2024-11-19 by pascal-roth