leggedrobotics/viplanner #37: [Question] When trying to run the demo for the Carla scene, IsaacSim gets stuck at this step and doesn't proceed.

revesreveilles, 2024-11-06:
```bash
./isaaclab.sh -p ~/viplanner/omniverse/standalone/viplanner_demo.py --scene carla --model_dir ~/viplanner/model
```
 Trying to run the `carla` demo,but the it gets stuck,and `isaacsim` doesn't load the Carla scene either.Not sure whhy it happened.
  I was running with the device and environment are listed below:
```bash
OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5
RTX3080 10G
Driver Version: 550.120  
cuda version: 12.1
PyTorch version: 2.2.2+cu121
```
[image]
And there is nothing on the simapp
[image]

revesreveilles, 2024-11-12:
somehow it works,but i have received the `torch dimension doesn't match `problem.
[image]
is it normal ?

pascal-roth, 2024-11-19:
Hi @revesreveilles 

First of all thanks for using our work. I tried to reproduce the error on my end, but everything worked as expected.

> Somehow it works,but i have received the `torch dimension doesn't match `problem. ![image](https://private-user-images.githubusercontent.com/169145696/385144350-2e81e2d2-093a-4527-9849-508fd8e415b7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwMDExNDUsIm5iZiI6MTczMjAwMDg0NSwicGF0aCI6Ii8xNjkxNDU2OTYvMzg1MTQ0MzUwLTJlODFlMmQyLTA5M2EtNDUyNy05ODQ5LTUwOGZkOGU0MTViNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExOVQwNzIwNDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNzRkYmRiNzI2OTdlZWQ5MTBhZmFmN2ZlNTZkNmY0NjkxNGVkMjI4ZTBkODYyMTIzM2NiN2U0YTFlYjcwOTIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zX3zrrY9QxBeSFOMlJfU_QGWLqSlVm83ljkLEmanezI) It did mentioned that `Mesh Town01_Opt has no faces or vertices` ![截图 2024-11-12 12-56-24](https://private-user-images.githubusercontent.com/169145696/385145137-83e5c5f6-4e08-4bb4-a3a2-741c4d6d2ab1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwMDExNDUsIm5iZiI6MTczMjAwMDg0NSwicGF0aCI6Ii8xNjkxNDU2OTYvMzg1MTQ1MTM3LTgzZTVjNWY2LTRlMDgtNGJiNC1hM2EyLTc0MWM0ZDZkMmFiMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExOVQwNzIwNDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lZWM3ZWMzZTk2OTJmZTZhMzJmMDk2YmQxMzAxOTI3ZmY0ZjM2OTQyNzI4MWE3YTA4ZTU1MjA5OTc4M2VmYjJiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.pdS0eDbNowLCAuNEx6fA88QmZ4GzPx-Qpi8Szc58_pM) Is it normal? The log file is as follows: [kit_20241111_201318.log](https://github.com/user-attachments/files/17712132/kit_20241111_201318.log)

This issue here looks very much like a wrong IsaacLab version to me. Can you ensure you are running IsaacSim 4.2 and IsaacLab version 1.2.0?  The triangle mesh cannot be generated for some landscape components, which shouldn't matter for the task. 

Let me know if you checked out the correct versions, then we can have a second look.

pascal-roth, 2024-11-23:
This seems to be the same issue as encountered in #38; please try their solution. In the case that it is still now working, please reopen this issue.

closed 2024-11-23 by pascal-roth