leggedrobotics/viplanner #38: viplanner isaaclab error

deavn2236, 2024-11-12:
I followed your instructions to configure isaaclab and viplanner, and wanted to try the effect under NVIDIA Warehouse. In the end, I received many warnings and isaaclab did not display. It is not a problem for me to open isaaclab normally without running the viplanner code. What is the reason for this? How can I solve it?
[image]

pascal-roth, 2024-11-19:
Hi @deavn2236 

First of all, thanks for using our code. 

You can ignore these warnings. The warehouse asset is downloaded from Nuclues, and that can take some time. Can you please ensure that you have a connection to Nuclues and then wait for a while to see if the asset is opened?

LTaitLeo, 2024-11-22:
I’m encountering exatly the same issue as described here. I can successfully connect to Nucleus, and all assets can be opened manually without any problem. However, when I run viplanner_demo, the process gets stuck here. Any assistance would be greatly appreciated.

LTaitLeo, 2024-11-22:
I’ve resolved the issue. It turns out that after installing the ViPlanner Omniverse Extension for the first time, a system restart and restarting the Omniverse Launcher were necessary before running the demo.

> I’m encountering exatly the same issue as described here. I can successfully connect to Nucleus, and all assets can be opened manually without any problem. However, when I run viplanner_demo, the process gets stuck here. Any assistance would be greatly appreciated.

pascal-roth, 2024-11-23:
@LTaitLeo thanks for finding a solution here, not sure why a restart is required.

closed 2024-11-23 by pascal-roth