leggedrobotics/viplanner #39: [Bug Report] Missing resampling_time_range in ViPlannerCarlaCfg

3rd-Musketeer, 2024-11-14:
### Checklist

- [x] I have checked that there is no similar issue in the repo
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo
- [x] I have tried to fix the bug myself, and it worked.

### Bug Description

The program throws an error about missing variable in `ViPlannerCarlaCfg`:
```shell
Traceback (most recent call last):
  File "/home/acir/viplanner/omniverse/standalone/viplanner_demo.py", line 151, in <module>
    main()
  File "/home/acir/viplanner/omniverse/standalone/viplanner_demo.py", line 71, in main
    env = ManagerBasedRLEnv(env_cfg)
  File "/home/acir/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 75, in __init__
    super().__init__(cfg=cfg)
  File "/home/acir/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 73, in __init__
    cfg.validate()
  File "/home/acir/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/utils/configclass.py", line 287, in _validate
    raise TypeError(
TypeError: Missing values detected in object ViPlannerCarlaCfg for the following fields:
  - commands.vel_command.resampling_time_range
```

### Steps to reproduce
I followed every step in the installation guidance and ensures that the IsaacSim/IsaacLab version complies with the requirements.
Then I ran this command to start the simulation in CARLA:
```
USER_PATH_TO_USD=~/viplanner/resources/new_carla_export ./isaaclab.sh -p ~/viplanner/omniverse/standalone/viplanner_demo.py --scene carla --model_dir ~/viplanner/models
```
(I changed `viplanner/omniverse/extension/omni.viplanner/omni/viplanner/config/carla_cfg.py` to accept environment variables as `USER_PATH_TO_USD`)
Then I got the error information above.

However, I tried to debug myself and located the missing variable at `IsaacLab/source/extensions/omni.viplanner/omni/viplanner/config/base_cfg.py` line 149, where this class lies:
```python
@configclass
class CommandsCfg:
    """Command specifications for the MDP."""

    vel_command: mdp.PathFollowerCommandGeneratorCfg = mdp.PathFollowerCommandGeneratorCfg(
        robot_attr="robot",
        lookAheadDistance=1.0,
        debug_vis=True,
        maxSpeed=1.0,
        # resampling_time_range=(4.0, 4.0),
    )
```
and I notice that for other config files in IsaacLab, they specify the `resampling_time_range` as a tuple, for example `(4.0, 4.0)`, and this tuple serves as the lower and upper boundary for a uniform sampling function, which is defined in `IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/command_manager.py` line 169, as:
```python
    def _resample(self, env_ids: Sequence[int]):
        """Resample the command.

        This function resamples the command and time for which the command is applied for the
        specified environment indices.

        Args:
            env_ids: The list of environment IDs to resample.
        """
        # resample the time left before resampling
        if len(env_ids) != 0:
            self.time_left[env_ids] = self.time_left[env_ids].uniform_(*self.cfg.resampling_time_range)
            # increment the command counter
            self.command_counter[env_ids] += 1
            # resample the command
            self._resample_command(env_ids)
```
To this end, I decided to specify a resample parameter manually, say `(4.0, 4.0)`, and the demo worked. 

I wonder if others encounter the same problem with me and shall we propose a PR to fix it? If so, what resample parameter should be used?

pascal-roth, 2024-11-19:
Hi @3rd-Musketeer 

Thanks a lot for using our work. The `_resample` function is called from the `compute` and `reset` method of the `CommandTerm`. Both should be overwritten in the `PathFollowerCommandGenerator`. 

Your issue is that you are in a newer IsaacLab version where the `Missing` check has been introduced. Please switch to version 1.2

pascal-roth, 2024-11-29:
Closing this issue for now, please reopen if it persists

closed 2024-11-29 by pascal-roth