leggedrobotics/viplanner #44: [Bug Report] Bug title ./isaaclab.sh -p <path-to-viplanner-repo>/omniverse/standalone/viplanner_demo.py find error

zhw970623, 2024-11-21:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

A clear and concise description of what the bug is.

When I verify the model demo, I execute cd IsaacLab
./isaaclab.sh -p <path-to-viplanner-repo>/omniverse/standalone/viplanner_demo.py --scene carla --model_dir <path-to-model-download-dir> and get an error

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

Describe the characteristic of your environment:

- Isaac Sim Version: [4.2.0]
- OS: [Ubuntu 20.04]
- GPU: [RTX 4070 Super]
- CUDA: [12.4]
- GPU Driver: [550.127.08.]

zhw970623, 2024-11-22:
viplanner 0.1.0 requires wandb==0.14.0,but you project need rl-games 1.6.1 , and  rl-games 1.6.1 need requires wandb<0.13.0,>=0.12.11
error message:
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
rl-games 1.6.1 requires wandb<0.13.0,>=0.12.11, but you have wandb 0.14.0 which is incompatible.

RROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
viplanner 0.1.0 requires wandb==0.14.0, but you have wandb 0.12.11 which is incompatible.

pascal-roth, 2024-11-22:
These pip errors should not influence the system if it normally installs. Did you try to run the demo?

zhw970623, 2024-11-22:
Yes, I tried the demo, and I allowed this command
./isaaclab.sh -p <path-to-viplanner-repo>/omniverse/standalone/viplanner_demo.py --scene carla --model_dir <path-to-model-download-dir>
ex:./isaaclab.sh -p /home/zhw/reach/baselineViplanner/viplanner/omniverse/standalone/viplanner_demo.py --scene carla --model_dir /home/zhw/reach/baselineViplanner/modle/model.pt

And I put usd_path="${USER_PATH_TO_USD}/carla.usd ---> in the file /home/zhw/reach/baselineViplanner/IsaacLab/source/extensions/omni.viplanner/omni/viplanner/config/carla_cfg.py usd_path="/home/zhw/reach/baselineViplanner/modle/new_carla_export/carla.usd"

Then the crash occurs, and the final error is as follows
`Traceback (most recent call last):
  File "/home/zhw/reach/baselineViplanner/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 64, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/home/zhw/reach/baselineViplanner/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 271, in _initialize_callback
    self._initialize_impl()
  File "/home/zhw/reach/baselineViplanner/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 388, in _initialize_impl
    import omni.replicator.core as rep
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/extscache/omni.replicator.core-1.11.20+106.1.0.lx64.r.cp310/omni/replicator/core/__init__.py", line 3, in <module>
    from .scripts import (
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/extscache/omni.replicator.core-1.11.20+106.1.0.lx64.r.cp310/omni/replicator/core/scripts/__init__.py", line 3, in <module>
    from . import (
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/extscache/omni.replicator.core-1.11.20+106.1.0.lx64.r.cp310/omni/replicator/core/scripts/backends/__init__.py", line 20, in <module>
    from .s3 import S3Backend
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/extscache/omni.replicator.core-1.11.20+106.1.0.lx64.r.cp310/omni/replicator/core/scripts/backends/s3.py", line 19, in <module>
    import boto3
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/exts/omni.pip.cloud/pip_prebundle/boto3/__init__.py", line 16, in <module>
    from boto3.compat import _warn_deprecated_python
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/exts/omni.pip.cloud/pip_prebundle/boto3/compat.py", line 19, in <module>
    from boto3.exceptions import PythonDeprecationWarning
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/exts/omni.pip.cloud/pip_prebundle/boto3/exceptions.py", line 15, in <module>
    import botocore.exceptions
ModuleNotFoundError: No module named 'botocore'
Traceback (most recent call last):
  File "/home/zhw/reach/baselineViplanner/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 64, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/home/zhw/reach/baselineViplanner/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 271, in _initialize_callback
    self._initialize_impl()
  File "/home/zhw/reach/baselineViplanner/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 388, in _initialize_impl
    import omni.replicator.core as rep
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/extscache/omni.replicator.core-1.11.20+106.1.0.lx64.r.cp310/omni/replicator/core/__init__.py", line 3, in <module>
    from .scripts import (
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/extscache/omni.replicator.core-1.11.20+106.1.0.lx64.r.cp310/omni/replicator/core/scripts/__init__.py", line 3, in <module>
    from . import (
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/extscache/omni.replicator.core-1.11.20+106.1.0.lx64.r.cp310/omni/replicator/core/scripts/backends/__init__.py", line 20, in <module>
    from .s3 import S3Backend
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/extscache/omni.replicator.core-1.11.20+106.1.0.lx64.r.cp310/omni/replicator/core/scripts/backends/s3.py", line 19, in <module>
    import boto3
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/exts/omni.pip.cloud/pip_prebundle/boto3/__init__.py", line 16, in <module>
    from boto3.compat import _warn_deprecated_python
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/exts/omni.pip.cloud/pip_prebundle/boto3/compat.py", line 19, in <module>
    from boto3.exceptions import PythonDeprecationWarning
  File "/home/zhw/.local/share/ov/pkg/isaac-sim-4.2.0/exts/omni.pip.cloud/pip_prebundle/boto3/exceptions.py", line 15, in <module>
    import botocore.exceptions
ModuleNotFoundError: No module named 'botocore'
2024-11-22 09:05:22 [24,876ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /Visuals/RayCaster.proto0_mesh_id0 that has not been populated
2024-11-22 09:05:22 [25,064ms] [Warning] [omni.hydra] Mesh '/World/envs/env_0/Robot/base/visuals.proto_mesh_0_id0' has corrupted data in primvar 'st': buffer size 702 doesn't match expected size 12828 in faceVarying primvars
[INFO]: Time taken for simulation start : 13.749385 seconds
[INFO] Command Manager:  <CommandManager> contains 1 active terms.
+----------------------------------------------------+
|                Active Command Terms                |
+-------+-------------+------------------------------+
| Index | Name        |             Type             |
[... cut ...]
  File "/home/zhw/reach/baselineViplanner/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 174, in data
    self._update_outdated_buffers()
  File "/home/zhw/reach/baselineViplanner/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 284, in _update_outdated_buffers
    outdated_env_ids = self._is_outdated.nonzero().squeeze(-1)
AttributeError: 'Camera' object has no attribute '_is_outdated'
2024-11-22 09:05:26 [28,836ms] [Warning] [carb] Recursive unloadAllPlugins() detected!`

zhw970623, 2024-11-22:
I also downloaded the model.yaml file, but I don't know which directory this file should be placed in

pascal-roth, 2024-11-23:
> I also downloaded the model.yaml file, but I don't know which directory this file should be placed in，but
> 
> When I execute `/isaaclab.sh -p /home/zhw/reach/baselineViplanner/viplanner/omniverse/standalone/viplanner_demo.py --scene carla --model_dir /home/zhw/reach/baselineViplanner/modle/model.pt `command, model.pt and modle.yaml are in the same folder

The '--model_dir' should be '/home/zhw/reach/baselineViplanner/modle'. Then both model.pt and model.yaml should be in there

pascal-roth, 2024-11-23:
> Yes, I tried the demo, and I allowed this command
>
> ./isaaclab.sh -p <path-to-viplanner-repo>/omniverse/standalone/viplanner_demo.py --scene carla --model_dir <path-to-model-download-dir> [...]

This looks to me like an error in your IsaacLab installation as replicator fails which has nothing to do with this code. Please redirect the issue there.

fan-ziqi, 2024-11-24:
Have you checked out the specified commit in the README?

IsaacLab 1.2.0 (commit: 0bccd886c19a9891c0b6bdd37e13f338eacc0bba)

pascal-roth, 2024-12-10:
Closing this issue now due to inactivity. 

Solution should be to checkout the correct commit

closed 2024-12-10 by pascal-roth