leggedrobotics/viplanner #52: [Question] problem about the model in the hardware depolyment!

command-z-z, 2024-11-30:
### Question
what's the difference of the following two lines of models?
- the first line is the models officially provided by mmdetection.
- the second line is the models downloading from your [readme](https://github.com/leggedrobotics/viplanner/blob/main/ros/INSTALL.md#models).
https://github.com/leggedrobotics/viplanner/blob/35a3bea47d423e1d15f38115cb6d1125f60b0f81/ros/planner/config/anymal_c.yaml#L11-L14
however, model architecture and weights appear inconsistent.

pascal-roth, 2024-12-07:
Hi @command-z-z ,

Thanks a lot for using our work. At the time when we did our experiments, the official weights did not fit to the code due to some renaming done in mmdetection, and we had to go some versions back to find the correct model, thus we included it here. Apparently, it is fixed now. You should be able to, therefore, use the latest official weights. 
I hope that answers your question.

command-z-z, 2024-12-08:
Think you for your reply, my concerns have been resolved.

closed 2024-12-08 by command-z-z