leggedrobotics/viplanner #55: [Question] /viplanner/.[inference,jetson]

AugWrite, 2024-12-10:
### Question

'The command '/bin/sh -c pip3 install --user -e /viplanner/.[inference,jetson]' returned a non-zero code: 1

this document doesn't have setup.py?

pascal-roth, 2024-12-10:
What is the error message?

AugWrite, 2024-12-11:
thinks！

AugWrite, 2024-12-11:
Thank you for your reply. I have understood the issue. However, I have encountered a new problem. I want to deploy a Docker environment and the viplanner project on a robotic dog. My version is r35.4.1. I tried using the `dustynv/l4t-pytorch:r35.4.1` image as the base image and executed the `build.sh` script inside it to install and configure ROS. However, it seems there are persistent issues with ROS.

After successfully deploying the new environment and running `catkin build viplanner_pkgs`, the following output appears. There are no errors, but the build does not proceed as expected.

I am unable to pinpoint the cause of the issue. Am I using the wrong base image, or is there another problem? 
[image]Subsequently, I tried other methods to install ROS1, but encountered errors.  “E: Sub-process /usr/bin/dpkg returned an error code (1)
Run CMD Task:[sudo apt install python3-rosdep -y]

pascal-roth, 2024-12-12:
Also inside the docker you have to create a catkin_ws. Basically execute the following commands before running the build process:

'''

mkdir -p ~/catkin_ws/src
source /opt/ros/noetic/setup.bash
cd ~/catkin_ws
catkin init
catkin config --extend /opt/ros/noetic
catkin config --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo

'''

pascal-roth, 2024-12-17:
Hi, 
did that fix your issue?

pascal-roth, 2024-12-26:
Hi @AugWrite,

Did you manage to solve your error and run ViPlanner on a different platform? Would be nice to get videos on the performance.

pascal-roth, 2025-01-09:
The issue is assumed to be fixed. Please reopen in case problems still exist.

closed 2025-01-09 by pascal-roth