leggedrobotics/viplanner #62: [Bug Report] Bug title

shiwentao12138, 2024-12-26:
Happy Christmas, I want to know why the path planning is so strange when using VIplanner
[image]

pascal-roth, 2024-12-26:
Hi @shiwentao12138, 

Thanks for using our work. 

I would need some more information to understand why ViPlanner is not performing as expected. 

- Can you show the input depth and semantic image? 
- What is the current orientation of the robot? 
- How is the goal provided to the robot? Is the visualisation in map/odom/base frame?

shiwentao12138, 2024-12-28:
Thank you for your response！

The input depth and semantic image:
[image]

The orientation of the robot is as shown in the coordinate system in the first picture：
[image]

And these are rosgraph and frames:
[image]
[image]

shiwentao12138, 2024-12-28:
Oh,sorry,the frames actually are: 
[image]

shiwentao12138, 2024-12-28:
And the model I used was downloaded from here：

[image]

pascal-roth, 2024-12-28:
> Thank you for your response！
> 
> The input depth and semantic image: ![2024-12-28 11-50-10 的屏幕截图](https://private-user-images.githubusercontent.com/147709540/399036442-2df5d7d5-65c7-41bc-a3f4-946cc9420999.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUzODczNzgsIm5iZiI6MTczNTM4NzA3OCwicGF0aCI6Ii8xNDc3MDk1NDAvMzk5MDM2NDQyLTJkZjVkN2Q1LTY1YzctNDFiYy1hM2Y0LTk0NmNjOTQyMDk5OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyOFQxMTU3NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMGExOTQwZTMwYmM3M2M3ZjU1MTM4ZTI3NjlkODExNDVkYzM5OTVlODA5NTk3NjBmZjc2YTYyYTg5Y2MzZjc3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.OLNqegfsqnoUYB0u-Tk8Ax8va2rTUnEVp8v-NhvjFtk)
> 
> The orientation of the robot is as shown in the coordinate system in the first picture： ![2024-12-26 17-07-11 的屏幕截图](https://private-user-images.githubusercontent.com/147709540/399036839-1c96f2e5-6c7f-4c43-bd05-5d9da40916c0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUzODczNzgsIm5iZiI6MTczNTM4NzA3OCwicGF0aCI6Ii8xNDc3MDk1NDAvMzk5MDM2ODM5LTFjOTZmMmU1LTZjN2YtNGM0My1iZDA1LTVkOWRhNDA5MTZjMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyOFQxMTU3NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZGMxNmFjMWFhZmM2YTc2ZDU2YzJkYTAzYjdhYTIwNWJiMzAwYWE5ZGIwNmM4MmRmMjhlOGZjNDY2N2IxNzI0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.kxDzGW0WtPbZVxMq-E2EIlfwbioj8_t-AfCWBoPZrgw)
> 
> And these are rosgraph and frames: ![frames](https://private-user-images.githubusercontent.com/147709540/399051040-0c8e9acf-dc9d-4ff7-a476-c79d69b999f0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUzODczNzgsIm5iZiI6MTczNTM4NzA3OCwicGF0aCI6Ii8xNDc3MDk1NDAvMzk5MDUxMDQwLTBjOGU5YWNmLWRjOWQtNGZmNy1hNDc2LWM3OWQ2OWI5OTlmMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyOFQxMTU3NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02OGM0MmEwMTQyZTkwMGM3Y2RkYTM5NzA0OGM2MzEzZTViMWQ2YmMwNjNiM2Q0YTQzNzdiYmJhYzM2NTAyNWVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.jMfPhzFGox1iLqj-xRWL5DDnJBfHyV7Wt0qkN5mvAYs) ![rosgraph](https://private-user-images.githubusercontent.com/147709540/399051042-1d8bdae6-5c47-475f-9097-6870adfbbaec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUzODczNzgsIm5iZiI6MTczNTM4NzA3OCwicGF0aCI6Ii8xNDc3MDk1NDAvMzk5MDUxMDQyLTFkOGJkYWU2LTVjNDctNDc1Zi05MDk3LTY4NzBhZGZiYmFlYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyOFQxMTU3NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hNDU4ODgwNTkyMWE0ZGY4M2VkNjZlNTU5YzkzYzAxNDkwNDNjZjBhNjlhNTQyMjE5Y2QyMGIzMjc5NjY3ZDUyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zKMWBxECUy2cWUBWjdeatZUxwvtbuNBRSIjwHqgAD08)

The semantic image seems wrong to me. Two things I noticed:
- The color of the ground is the color of static obstacles (e.g., a wall or a building)
- There is also a black (i.e., high-risk) region in front of the robot. The planner has been trained with a camera view that does not include the robot itself. This is an out-of-distribution setting. 

Given those two, there is not a clear path to take which can exlain the weird paths that you are experiencing.

shiwentao12138, 2024-12-30:
Thank you for your answer. I have learned a lot.