leggedrobotics/viplanner #63: "The model and loaded state dict do not match exactly."

shiwentao12138, 2024-12-30:
I encountered this problem while using the pre training file you recommended to download for Mask2Former semantic segmentation: "The model and loaded state dict do not match exactly."

So I used this link :"https://download.openmmlab.com/mmdetection/v3.0/mask2former/mask2former_r50_8xb2-lsj-50e_coco-panoptic/mask2former_r50_8xb2-lsj-50e_coco-panoptic_20230118_125535-54df384a.pth"  to download pre training files. 
Is this suitable for VIPlaner?

shiwentao12138, 2024-12-30:
[image]
I'm not sure if there's an issue with the pre-trained file you provided. If there's a problem, could you provide a new and suitable pre-trained file?

pascal-roth, 2024-12-31:
Hi, 

Please always check the existing issues. This should be fixed with the correct versions mentioned in #23

pascal-roth, 2025-01-09:
The issue is assumed to be fixed. Please reopen in case problems still exist.

closed 2025-01-09 by pascal-roth