leggedrobotics/viplanner #64: [Question] Data_collect_error (matterport)

blueblack649, 2025-01-02:
### Question
I successfully modified the Matterport USD file. After that, I tried to extract data using `data_collect.py`. For the `.ply` file, I used the one located in the `House_segmentations` directory. I did not modify any part of the existing code except for setting the path.

However, I am continuously encountering an issue where the `MatterportRayCasterCamera` reports that it does not have the attribute `meshes`. I am unsure how to resolve this problem. Could you help me?

### matterport_cfg.py

```
# Copyright (c) 2023-2024, ETH Zurich (Robotics Systems Lab)
# Author: Pascal Roth
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

import omni.isaac.lab.sim as sim_utils
import omni.viplanner.viplanner.mdp as mdp
from omni.isaac.lab.assets import AssetBaseCfg
from omni.isaac.lab.managers import ObservationGroupCfg as ObsGroup
from omni.isaac.lab.managers import ObservationTermCfg as ObsTerm
from omni.isaac.lab.managers import SceneEntityCfg
from omni.isaac.lab.scene import InteractiveSceneCfg
from omni.isaac.lab.sensors import ContactSensorCfg, patterns
from omni.isaac.lab.utils import configclass
from omni.isaac.matterport.config import MatterportImporterCfg
from omni.isaac.matterport.domains import MatterportRayCasterCfg
from omni.viplanner.utils import VIPlannerMatterportRayCasterCameraCfg

from .base_cfg import ObservationsCfg, ViPlannerBaseCfg

##
# Pre-defined configs
##
# isort: off
from omni.isaac.lab_assets.anymal import ANYMAL_C_CFG

##
# Scene definition
##

@configclass
class TerrainSceneCfg(InteractiveSceneCfg):
    """Configuration for the terrain scene with a legged robot."""

    # ground terrain
    terrain = MatterportImporterCfg(
        prim_path="/World/matterport",
        terrain_type="matterport",
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
        ),
        # NOTE: this path should be absolute to load the textures correctly
        # obj_filepath="${USER_PATH_TO_USD}/matterport.usd",
        obj_filepath="/home/jong/IsaacLab/viplanner/omniverse/extension/omni.isaac.matterport/data/mappings/2azQ1b91cZZ/matterport_mesh/2azQ1b91cZZ/matterport_mesh/7812e14df5e746388ff6cfe8b043950a/a7812e14df5e746388ff6cfe8b043950a.usd",
        groundplane=True,
    )
    # robots
    robot = ANYMAL_C_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")
    robot.init_state.pos = (8.0, 0.0, 0.6)
    robot.init_state.rot = (0.6126, 0.0327, 0.0136, -0.7896)

    # sensors
    height_scanner = MatterportRayCasterCfg(
        prim_path="{ENV_REGEX_NS}/Robot/base",
        offset=MatterportRayCasterCfg.OffsetCfg(pos=(0.0, 0.0, 0.5)),
        attach_yaw_only=True,
        pattern_cfg=patterns.GridPatternCfg(resolution=0.1, size=[1.6, 1.0]),
        debug_vis=True,
[... 272 lines cut ...]
    self._initialize_impl()
  File "/home/jong/IsaacLab/viplanner/omniverse/extension/omni.isaac.matterport/omni/isaac/matterport/domains/matterport_raycast_camera.py", line 59, in _initialize_impl
    super()._initialize_impl()
  File "/home/jong/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/ray_caster/ray_caster.py", line 147, in _initialize_impl
    self._initialize_warp_meshes()
  File "/home/jong/IsaacLab/viplanner/omniverse/extension/omni.isaac.matterport/omni/isaac/matterport/domains/matterport_raycast_camera.py", line 82, in _initialize_warp_meshes
    mesh_prim_path in MatterportRayCasterCamera.meshes
AttributeError: type object 'MatterportRayCasterCamera' has no attribute 'meshes'
[INFO]: Time taken for simulation start : 0.397843 seconds
[INFO]: Setup complete...
[INFO] No viewpoint samples found for seed 1 and 10000 samples.
[INFO] Starting terrain analysis...
Traceback (most recent call last):
  File "/home/jong/IsaacLab/viplanner/omniverse/standalone/data_collect.py", line 134, in <module>
    main()
  File "/home/jong/IsaacLab/viplanner/omniverse/standalone/data_collect.py", line 116, in main
    samples = explorer.sample_viewpoints(args_cli.num_samples)
  File "/home/jong/IsaacLab/viplanner/omniverse/extension/omni.viplanner/omni/viplanner/collectors/viewpoint_sampling.py", line 66, in sample_viewpoints
    self.terrain_analyser.analyse()
  File "/home/jong/IsaacLab/viplanner/omniverse/extension/omni.viplanner/omni/viplanner/collectors/terrain_analysis.py", line 72, in analyse
    self._setup_raycaster()
  File "/home/jong/IsaacLab/viplanner/omniverse/extension/omni.viplanner/omni/viplanner/collectors/terrain_analysis.py", line 704, in _setup_raycaster
    if isinstance(self._raycaster.meshes[self._raycaster.cfg.mesh_prim_paths[0]], list):
KeyError: '/home/jong/IsaacLab/viplanner/omniverse/extension/omni.isaac.matterport/data/mappings/2azQ1b91cZZ/house_segmentations/2azQ1b91cZZ/house_segmentations/2azQ1b91cZZ.ply'
```

pascal-roth, 2025-01-02:
Hi, 

Thanks for using our work. This is very likely a version issue of IaaacLab. Please make sure that you checkout version 1.2 (even better the comment mentioned in the README.md).

Please let us know if that fixes your issue.

pascal-roth, 2025-01-09:
The issue is assumed to be fixed. Please reopen in case problems still exist.

closed 2025-01-09 by pascal-roth