leggedrobotics/viplanner #68: [Bug Report] TypeError: Missing values detected in object ViPlannerCarlaCfg for the following fields:   - commands.vel_command.resampling_time_range

lfsy147, 2025-01-09:
### Describe the bug
Hi,
I followed your instructions to configure isaaclab and viplanner, and wanted to try the effect under carla. In the end, I received many warnings and isaaclab did not display. How can I solve it?

### Steps to reproduce

After I run this code ,I receive the fllow error report. I make sure  the path I entered includes the .pt model file and the .ymal configuration file.
```
./isaaclab.sh -p /home/fsy/code/viplanner/omniverse/standalone/viplanner_demo.py --scene carla --model_dir /home/fsy/code/viplanner/model
```

```
Traceback (most recent call last):
  File "/home/fsy/code/viplanner/omniverse/standalone/viplanner_demo.py", line 151, in <module>
    main()
  File "/home/fsy/code/viplanner/omniverse/standalone/viplanner_demo.py", line 71, in main
    env = ManagerBasedRLEnv(env_cfg)
  File "/home/fsy/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 76, in __init__
    super().__init__(cfg=cfg)
  File "/home/fsy/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 77, in __init__
    cfg.validate()
  File "/home/fsy/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/utils/configclass.py", line 287, in _validate
    raise TypeError(
TypeError: Missing values detected in object ViPlannerWarehouseCfg for the following fields:
  - commands.vel_command.resampling_time_range

```
[image]

-->

lfsy147, 2025-01-09:
https://github.com/leggedrobotics/viplanner/issues/44

closed 2025-01-09 by lfsy147

XanderWangXY, 2025-05-09:
Hi! How did you solve this problem?

lfsy147, 2025-05-09:
> Hi! How did you solve this problem?
If you use the latest IsaacSim and IsaacLan follow this issue
https://github.com/leggedrobotics/viplanner/issues/39  or follow https://github.com/leggedrobotics/viplanner/issues/44