leggedrobotics/viplanner #73: [Question] Question title

eamon2003, 2025-01-18:
### Question
how can i change the default robot to another one in NIVIDA warehouse environment

pascal-roth, 2025-01-18:
Hi,

You can change that in the config at line https://github.com/leggedrobotics/viplanner/blob/main/omniverse/extension/omni.viplanner/omni/viplanner/config/warehouse_cfg.py#L51. Be aware that also the locomotion policy in the navigation actions have to be adjusted.

eamon2003, 2025-01-22:
> Hi, 你好,
> 
> You can change that in the config at line https://github.com/leggedrobotics/viplanner/blob/main/omniverse/extension/omni.viplanner/omni/viplanner/config/warehouse_cfg.py#L51. Be aware that also the locomotion policy in the navigation actions have to be adjusted.您可以在配置文件的第 51 行进行更改，该行位于 https://github.com/leggedrobotics/viplanner/blob/main/omniverse/extension/omni.viplanner/omni/viplanner/config/warehouse_cfg.py#L51 。请注意，导航动作中的运动策略也需要进行调整。

thank you!

closed 2025-01-25 by pascal-roth