leggedrobotics/viplanner #76: [Question] cannot acess cloud.ply

leo7611, 2025-02-18:
### Question

I apologize for the interruption, but I'm really eager to know how I can obtain the "cloud.ply" file. It's of great significance to me, and I'd greatly appreciate your guidance.

Traceback (most recent call last):
  File "/home/xxx/viplanner_data_generator/standalone/omni.viplanner.collectors/check_carla_trajectory_sampling.py", line 32, in <module>
    from omni.viplanner.collectors.collectors import (
ModuleNotFoundError: No module named 'omni.viplanner.collectors.collectors'
2025-02-18 11:45:39 [18,524ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,524ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,524ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,524ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,524ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,524ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,524ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,524ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,525ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,525ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,525ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,525ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-18 11:45:39 [18,525ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2025-02-18 11:45:39 [18,643ms] [Warning] [carb] Recursive unloadAllPlugins() detected!

pascal-roth, 2025-02-18:
Hi @leo7611,

Thanks for using our work. 

The issue seems to be related to a wrong installation as the package cannot be found. Please, see again the installation guide and make sure you follow every step and have the correct versions for IsaacLab and IsaacSim.

leo7611, 2025-02-20:
> Hi [@leo7611](https://github.com/leo7611),
> 
> Thanks for using our work.
> 
> The issue seems to be related to a wrong installation as the package cannot be found. Please, see again the installation guide and make sure you follow every step and have the correct versions for IsaacLab and IsaacSim.

Thank you for your reply. I'm using the code in this repo: https://github.com/fan-ziqi/viplanner_data_generator.git. When I run it following the steps in this repo, it reports an error: `from omni.viplanner.collectors.collectors import (
ModuleNotFoundError: No module named 'omni.viplanner.collectors.collectors'` I noticed that it installed "orbit-nav-collectors" instead of "omni-viplanner-collectors".

leo7611, 2025-02-20:
> Thank you for your reply. I'm using the code in this repo: https://github.com/fan-ziqi/viplanner_data_generator.git. When I run it following the steps in this repo, it reports an error: `from [...]

How can I obtain the `cloud.ply` file of Carla Town01 in the example routine? Just like the training set in the picture in [https://github.com/leggedrobotics/viplanner/issues/51](https://github.com/leggedrobotics/viplanner/issues/51), I'm only missing the `cloud.ply` file. I originally thought I needed to follow the steps in [https://github.com/fan-ziqi/viplanner_data_generator.git](https://github.com/fan-ziqi/viplanner_data_generator.git) step by step, but such a problem has occurred. 

[image]

pascal-roth, 2025-02-20:
This is a repo not supported by the authors of this work. Simply use the generation code provided in this repo https://github.com/leggedrobotics/viplanner/blob/main/omniverse/standalone/data_collect.py and then execute the script to reconstruct the cost

leo7611, 2025-02-20:
> /

Thank you for your reply. I used **data_collect.py** to obtain data and got **viewpoints_seed1_samples10000.pkl**, but the **cloud.ply** file is missing. This causes the following error when I use the Town01 data of Carla in the **cost-builder**:

`python viplanner/cost_builder.py
============ Creating Semantic Map from cloud ===============
COST-MAP INIT START
start loading and filtering point cloud from: cloud.ply
Traceback (most recent call last):
  File "/home/xxx/viplanner/viplanner/cost_builder.py", line 51, in <module>
    main(cfg)
  File "/home/xxx/viplanner/viplanner/cost_builder.py", line 19, in main
    sem_cost_map.pcd_init()
  File "/home/xxx/miniconda3/envs/viplanner/lib/python3.10/site-packages/viplanner/cost_maps/sem_cost_map.py", line 66, in pcd_init
    assert os.path.exists(pc_path), f"point cloud file does not exist: {pc_path}"
AssertionError: point cloud file does not exist: <path-to-data>/<env-name>/cloud.ply`

pascal-roth, 2025-02-20:
You have to adjust the config to your file location, as described in the READMEs! Please check these first

leo7611, 2025-02-20:
> You have to adjust the config to your file location, as described in the READMEs! Please check these first

I'm very sorry to bother you. I carefully read the readme.md again and then ran data_collect.py once more:  ```
./isaaclab.sh -p /home/xxx/viplanner/omniverse/standalone/data_collect.py --scene carla --num_samples 10000```，
The log is as shown below. However, the semantics folder is empty while the depth and rgb folders are normal：

```
[INFO] Loaded 10000 with seed 1.
[INFO] Start rendering 10000 images.
[INFO] Saving camera configurations to /home/xxx/viplanner/usd/carla_export/carla.
2025-02-20 07:22:15 [83,582ms] [Warning] [omni.physx.cooking.plugin] m_processResult == OperationResult::SUCCESS && "result parameter" is false.
2025-02-20 07:22:15 [83,582ms] [Warning] [omni.physx.cooking.plugin] UjitsoMeshCookingContext: cooking failure for /World/Carla/terrain/Town01_Opt/LandscapeComponent_13
2025-02-20 07:22:17 [85,617ms] [Warning] [omni.physx.cooking.plugin] m_processResult == OperationResult::SUCCESS && "result parameter" is false.
2025-02-20 07:22:17 [85,618ms] [Warning] [omni.physx.cooking.plugin] UjitsoMeshCookingContext: cooking failure for /World/Carla/terrain/Town01_Opt/LandscapeComponent_30
2025-02-20 07:22:19 [87,392ms] [Warning] [omni.physx.cooking.plugin] m_processResult == OperationResult::SUCCESS && "result parameter" is false.
2025-02-20 07:22:19 [87,392ms] [Warning] [omni.physx.cooking.plugin] UjitsoMeshCookingContext: cooking failure for /World/Carla/terrain/Town01_Opt/LandscapeComponent_20
[INFO] Rendered 100 images in 43.0057s.
[INFO] Rendered 200 images in 75.1756s.
[INFO] Rendered 300 images in 105.1648s.
。。。。。
[INFO] Rendered 19800 images in 5799.2311s.
2025-02-20 08:59:04 [5,892,729ms] [Warning] [omni.kit.window.property.templates.simple_property_widget] CameraSchemaAttributesWidget.build_items took 0.44448723102686927 seconds
[INFO] Rendered 19900 images in 5827.6295s.
[INFO] Rendered 20000 images in 5855.9548s.
[INFO]: Viewpoints sampled.
Rendering will continue to render the environment and visualize the last camera positions.
```

Due to the lack of contents in the carla_export/carla/semantics folder, I'm unable to perform the depth_reconstruct. The running log is as follows:

```
python viplanner/depth_reconstruct.py 

Ready to read depth data.
total number of images for reconstruction: 10000
Reconstructing 3D Points:   0%|                       | 0/10000 [00:00<?, ?it/s][ WARN:0@4.072] global loadsave.cpp:248 findDecoder imread_('/home/xxx/viplanner/usd/carla_export/carla/semantics/0000.png'): can't open/read file: check file path/integrity
Reconstructing 3D Points:   0%|                       | 0/10000 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "/home/xxx/viplanner/viplanner/depth_reconstruct.py", line 367, in <module>
    depth_constructor.depth_reconstruction()
  File "/home/xxx/viplanner/viplanner/depth_reconstruct.py", line 139, in depth_reconstruction
    sem_annotation, filter_idx = self._get_semantic_image(points_final, img_idx)
  File "/home/xxx/viplanner/viplanner/depth_reconstruct.py", line 332, in _get_semantic_image
    sem_image = cv2.cvtColor(sem_image, cv2.COLOR_BGR2RGB)
cv2.error: OpenCV(4.9.0) /io/opencv/modules/imgproc/src/color.cpp:196: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
```

pascal-roth, 2025-02-20:
Did you check that you are using the correct version of IsaacSim and IsaacLab? I checked it on my side and the semantics are rendered, thus I cannot replicate the behavior, especially since there is no error message for the semantics.

leo7611, 2025-02-20:
> Did you check that you are using the correct version of IsaacSim and IsaacLab? I checked it on my side and the semantics are rendered, thus I cannot replicate the behavior, especially since there is no error message for the semantics.

I really want to figure out the reason, so I checked again. The versions I'm using are indeed the recommended ones,  `IsaacSim 4.2.0 and IsaacLab 1.2.0 (commit: 0bccd886c19a9891c0b6bdd37e13f338eacc0bba)`. 
Moreover, the only change I made was to the path in the file`/home/xxx/viplanner/omniverse/extension/omni.viplanner/omni/viplanner/config/carla_cfg.py`, 
which is `usd_path="/home/xxx/viplanner/data/new_carla_export/carla.usd"`. 
The collect command I used is`./isaaclab.sh -p /home/xxx/viplanner/omniverse/standalone/data_collect.py --scene carla --num_samples 2000`.

closed 2025-03-07 by leo7611