leggedrobotics/viplanner #77: [Question] How to get data?

Un1Lee, 2025-02-20:
### Question

As TRAINING.md mentioned, I get scans data 2azQ1b01cZZ, including cameras.zip, matterport_camera_intrinsics.zip, matterport_camera_images.zip, matterport_mesh.zip, matterport_color_images.zip, matterport_depth_images.zip, regin_segmentation.zip, house_segmentations.zip and so on.

And I unzip matterport_mesh.zip, and convert the .obj to .usd. But when I want to get the cost map, I am confused about which zip the depth and semantics from?  Whatever, there has usd, no need to depth reconstruction?

Anyway, the queation is how to get the data which cost-buiding needs from my zip download. such as depth xxx.png, xxx.npy.

pascal-roth, 2025-02-20:
For the semantics use the ply file in 'house_segmentations.zip', then adjust the necessary configs and execute https://github.com/leggedrobotics/viplanner/blob/main/omniverse/standalone/data_collect.py to collect the data (depth and semantic images)

Un1Lee, 2025-02-21:
I have get cost map successully . Thank you! Anyway, can I ask what number of num_samples should be? I use 2000, ok?

pascal-roth, 2025-02-21:
2000 should be sufficient, the more images the less unobstructed spots and the better the cost map

closed 2025-02-21 by pascal-roth

leo7611, 2025-02-21:
> I have get cost map successully . Thank you! Anyway, can I ask what number of num_samples should be? I use 2000, ok?

May I ask how you obtained the **semantic** map and the **.ply** file? Why can't I get the corresponding files even though my code runs normally?https://github.com/leggedrobotics/viplanner/issues/76