leggedrobotics/viplanner #78: [Question] beat_the_planner_cfg does not exists

Jaehwi-Song, 2025-02-21:
### Question

I'm trying to test the functions of "feature/beat_the_planner" branch. However, there are no beat_the_planner_cfg folder in omni.viplanner.config directory.

# viplanner_demo.py, line 52 cannot be imported
from omni.viplanner.config.beat_the_planner_cfg import (
    BeatThePlannerCarlaCfg,
    BeatThePlannerMatterportCfg,
    BeatThePlannerWarehouseCfg,
)

ModuleNotFoundError: No module named 'omni.viplanner.config.beat_the_planner_cfg'

Is there any way that I can get those config files?

pascal-roth, 2025-02-21:
Hi, 

Thanks for using our work. 

This is not an official supported feature and was used for a demo. I can see if I still find the configs somewhere

pascal-roth, 2025-07-08:
Configs are now uploaded in the branch, see here https://github.com/leggedrobotics/viplanner/blob/5bf5f1008fe4a06d7db55e1d169dac6bbbb2442b/omniverse/extension/omni.viplanner/omni/viplanner/config/beat_the_planner_cfg.py