leggedrobotics/viplanner #79: Does the trained model you provided adapt to the real world or other virtual environments?

shiwentao12138, 2025-02-22:
Hi, does the trained model you provided (as shown below) adapt to the real world or other virtual environments？
[image]

Because when I use VIPlanner for navigation, the route it plans does not respond to obstacles in the virtual indoor environment(as shown below).
[image]
The robot will follow the planned route and collide with the wall.

pascal-roth, 2025-02-22:
Hi, 

Thanks for using our work. 

The model should translate to the real world. It is very sensitive to the camera and its placement, so please check that the camera intrinsic and extrinsic match with the ones it has been trained with. Moreover, it is possible that the path at some point goes into the wall but when iteratively replanning (ideally 10 hz), it should not collide.

pascal-roth, 2025-03-21:
closing this issue now due to inactivity

closed 2025-03-21 by pascal-roth

Leowu-git, 2025-04-23:
Hi, I got the trained model you provided, then how should I make sure the camera intrinsic and extrinsic matched with the ones it has been trained with? I don't find the way to get camera information of  trained model.
I really need your help, thank you!