leggedrobotics/viplanner #81: [Question] How to get the Goal  Reached in test?

Un1Lee, 2025-03-03:
How to get the data like your TABLE 2 in your paper? In testing, it only reports val_loss and test_loss. How can I get Goal  Reached. Anyway, I can not get Geom. Loss and Sem. Loss, too.

Un1Lee, 2025-03-09:
I really need your help!

pascal-roth, 2025-03-13:
Hi @Un1Lee 

the code for that evaluation was never cleaned and is on a very old version of IsaacLab (prev. Oribit). I can share you the files, but you would have to reimplment them for the latest changes

Un1Lee, 2025-03-21:
> Hi [@Un1Lee](https://github.com/Un1Lee)
> 
> the code for that evaluation was never cleaned and is on a very old version of IsaacLab (prev. Oribit). I can share you the files, but you would have to reimplment them for the latest changes

It's ok. Can you send it to my email(uni_lee@outlook.com)?