leggedrobotics/viplanner #84: [Question] how to change the assets path

eamooooon, 2025-03-19:
i want to change the assets path to local path, then I change the omniverse‘s user.config.json "persistent/isaac/isaac/default" to my local path ”/home/eai-amd/lrt/IsaacAssets"
then i thought that was ok and i put my robot in my assets path.
when I tried to start it, it had produced an error report
```
Traceback (most recent call last):
  File "/home/eai-amd/lrt/viplanner/omniverse/standalone/viplanner_demo.py", line 151, in <module>
    main()
  File "/home/eai-amd/lrt/viplanner/omniverse/standalone/viplanner_demo.py", line 71, in main
    env = ManagerBasedRLEnv(env_cfg)
  File "/home/eai-amd/lrt/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 75, in __init__
    super().__init__(cfg=cfg)
  File "/home/eai-amd/lrt/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 116, in __init__
    self.scene = InteractiveScene(self.cfg.scene)
  File "/home/eai-amd/lrt/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/interactive_scene.py", line 145, in __init__
    self._add_entities_from_cfg()
  File "/home/eai-amd/lrt/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/interactive_scene.py", line 452, in _add_entities_from_cfg
    self._articulations[asset_name] = asset_cfg.class_type(asset_cfg)
  File "/home/eai-amd/lrt/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 99, in __init__
    super().__init__(cfg)
  File "/home/eai-amd/lrt/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/asset_base.py", line 82, in __init__
    self.cfg.spawn.func(
  File "/home/eai-amd/lrt/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/utils.py", line 245, in wrapper
    prim = func(prim_paths[0], cfg, *args, **kwargs)
  File "/home/eai-amd/lrt/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/spawners/from_files/from_files.py", line 60, in spawn_from_usd
    return _spawn_from_usd_file(prim_path, cfg.usd_path, cfg, translation, orientation)
  File "/home/eai-amd/lrt/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/spawners/from_files/from_files.py", line 222, in _spawn_from_usd_file
    raise FileNotFoundError(f"USD file not found at path: '{usd_path}'.")
FileNotFoundError: USD file not found at path: 'http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2/Isaac/Robots/Humanoid/humanoid_publish.usd'.
```
I found that this usd_path has not be change totally
so how can I change it to my local path 
thanks

pascal-roth, 2025-03-21:
Hi @eamooooon 

Thanks for using our work. To change the robot is sadly not so trivial as just changing a path. You will have to change the articulation config your humanoid which then also includes the path to your asset (you can put there an absolute path). For the Carla example, this should be done here: https://github.com/leggedrobotics/viplanner/blob/main/omniverse/extension/omni.viplanner/omni/viplanner/config/carla_cfg.py#L52-L53

In addition, you will have to modify the locomotion policy as this one is only valid for ANYmal (define the adjusted observation space and change the policy file). 

If you manage to apply the planner on a humanoid, it would be very cool to get a video of it and maybe have the configs available to the public.

pascal-roth, 2025-03-21:
The absolute path should be correct. Can you copy the entire terminal output and see if there is any error above that? It seems that the robot is not correctly loaded and then the camera prim is not found.

pascal-roth, 2025-03-21:
When you open your usd asset, the base path exists? If the asset is correct, then I don't really know what could be the issue ... should work.

eamooooon, 2025-04-02:
> When you open your usd asset, the base path exists? If the asset is correct, then I don't really know what could be the issue ... should work.

I solve this question finally. 
The next question is how can I change the locomotion policy to my own. 
I already have a locomotion policy, but it is base on mujoco not Isaaclab
so I will try to change it to fit in your code
thank you

pascal-roth, 2025-04-02:
great, good luck with the next step

closed 2025-04-02 by pascal-roth

eamooooon, 2025-04-03:
> great, good luck with the next step太好了，祝你下一步顺利。

Could you please tell me how to change the locomotion of your code?