leggedrobotics/viplanner #85: [Question] Question title

jmk7791, 2025-03-20:
### Question

Hi, I would like to use ViPlanner ROS Node in CARLA Town10 with person ego vehicle (using "roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch"). 

I am using CARLA's segmenatation topic "/carla/ego_vehicle/semantic_segmentation_front/image" instead of mask2former. (I modified the color of segments same as viplanner). I downloaded the model you provide. I put model.pt and model.yaml in "ros/planner/models" path. 

However, when I launched roslaunch viplanner_node viplanner.launch rviz:=true, the ego_vehicle follows the goal, but cannot avoid the obstacles, and do not follow the semantic info. (ex. It does not follow the crosswalk, and just walk through the road.) 

Can you guess what is the problem? Or is there any additional thing that I should do to use viplanner ros node in CARLA? 
Below is my anymal_mount.yaml. 
Thanks for your kind support. 

[image]

pascal-roth, 2025-03-21:
Hi @jmk7791 

Thanks for using our work. I personally don't know the Carla Person Ego Vehicle, is it also omnidirectional? 

There are two things to check:
1. Are the camera intrinsics and extrinsics the same? The policy is very sensitive to those 
2. Is the Color mapping correct? 

The current policy also has certain limitations. It has a pretty strong goal attraction, which means that if there is no crosswalk close, it will go over the road to go to the goal. This is getting stronger if the goal is further away, as only 5 waypoints are produced by the policy. Do you maybe have an example image for the use case where it fails?

jmk7791, 2025-03-24:
Thanks for your reply. 
1. the camera I'm using (CARLA semantic segmentation camera) intrinsics and extrinsic are like below since my json file is like this. 

[image]

[image]

2. I mapped the color same as Viplanner.

[image]

3. This is my example with ego vehicle (pedestrian, the yellow person in the picture) near the crosswalk. It didn't follow the crosswalk.  Plus, I cannot visualize the goal and path in the rviz. How can I visualize it? I launched the viplannerViz.launch file and add /viplanner/viz_path_rgb_image, I cannot see any goal or path visualized. 
4. 

[image]

pascal-roth, 2025-04-15:
Is this issue resolved?