leggedrobotics/viplanner #87: [Question] issue with IsaacLab versions

chengweialan, 2025-03-26:
I noticed that your code is updated to a new IsaacLab version. Before it was tested on 1.2.0. Is the code still compatible for this 1.2.0 version? Are there any other future versions supported? 

I used IsaacLab 1.4.0 to avoid dependency conflicts, but encountered with following issues:

> Traceback (most recent call last):
  File "/home/linux/Desktop/Lab1.4/IsaacLab-1.4.0/../../viplanner/viplanner/omniverse/standalone/viplanner_demo.py", line 151, in <module>
    main()
  File "/home/linux/Desktop/Lab1.4/IsaacLab-1.4.0/../../viplanner/viplanner/omniverse/standalone/viplanner_demo.py", line 71, in main
    env = ManagerBasedRLEnv(env_cfg)
  File "/home/linux/Desktop/Lab1.4/IsaacLab-1.4.0/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 76, in __init__
    super().__init__(cfg=cfg)
  File "/home/linux/Desktop/Lab1.4/IsaacLab-1.4.0/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 77, in __init__
    cfg.validate()
  File "/home/linux/Desktop/Lab1.4/IsaacLab-1.4.0/source/extensions/omni.isaac.lab/omni/isaac/lab/utils/configclass.py", line 287, in _validate
    raise TypeError(
TypeError: Missing values detected in object ViPlannerWarehouseCfg for the following fields:
   commands.vel_command.resampling_time_range

Besides, has the code been tested for new IsaacSim 4.5.0? I wonder if any of you have implemented it on IsaacSim 4.5.0 and any other later IsaacLab versions.

pascal-roth, 2025-03-26:
Hi, 

The code has not been updated to any new version. It is only tested with IsaacSim 4.2 and IsaacLab 1.2; please see the README for details on the commit. There will be no more updates from our side; however, if you updated the codebase for a newer version, please make a PR, as this would be a great addition.

pascal-roth, 2025-04-15:
Assuming this issue has been resolved with the version update, closing it now

closed 2025-04-15 by pascal-roth