leggedrobotics/viplanner #91: [Question] where loaded low-level locomotion model?

FanMecha, 2025-04-11:
Hi, I have already built the cost map and successfully trained the ViPlanner model. However, when I run viplanner_demo, I don't see where the pre-trained locomotion control policy or navigation model is being loaded.

My understanding is that after predicting a trajectory, there should be a model that tracks this trajectory and outputs velocity commands, which are then passed to the low-level locomotion controller. I want to know where exactly these two policies are being loaded.

FanMecha, 2025-04-15:
Thank you for your wonderful work. We retrained the network and deployed it on my humanoid robot with great results. so cool

pascal-roth, 2025-04-21:
Closing this issue for now. If you have any material, it would be great if you can forward it to rothpa@ethz.ch

closed 2025-04-21 by pascal-roth

FanMecha, 2025-04-22:
> This is great news, could you share some videos/ images or anything? That would be very nice to include also here in the repo

of course ,we replace anymal-c to my humanoid, and we try to deploy in real world.

[image]

pascal-roth, 2025-04-22:
very cool, if you have your work somewhere, we can make a link to your repo to promote it (seems like a very cool addition)

Dream3Lab, 2025-05-25:
Thank you for your great work. I would like to also ask that where is the pretrained locomotion model loaded. Also, is the locomotion model specifically designed for ANYmal robot?