leggedrobotics/viplanner #92: [Question] Question title

lrt20021018, 2025-04-19:
when i run
`./isaaclab.sh -p /home/lrt/robot_asset/viplanner/omniverse/standalone/viplanner_demo.py --scene warehouse --model_dir /home/lrt/robot_asset/model`
there are lots of errors cycle like
```
2025-04-19 08:39:17 [262,164ms] [Error] [rtx.materialdb.plugin] Texture upload failed for texture omniverse://localhost/NVIDIA/Assets/Isaac/2023.1.0/Isaac/Environments/Simple_Warehouse/Materials/Textures/T_AisleSign_N.png
2025-04-19 08:39:17 [262,165ms] [Error] [rtx.materialdb.plugin] Failed texture loads reported for material M_AisleSign(texture_2d)
2025-04-19 08:39:17 [262,165ms] [Error] [omni.hydra] Failed wait for MDL material (M_AisleSign(texture_2d)) snapshot
```
how can i solve this problem
thank you !

pascal-roth, 2025-04-21:
Hi,
Thanks for using our work. The warehouse asset is referenced to 'localhost' which existed in prev. versions when downloading the omnivores launcher and installing nucleus in there. This should also make it work for you. Better solution is to update all the asset paths and directly reference the NVIDIA could location.

lrt20021018, 2025-04-21:
> Hi,  你好 Thanks for using our work. The warehouse asset is referenced to 'localhost' which existed in prev. versions when downloading the omnivores launcher and installing nucleus in there. This should also make it work for you. Better solution is to update all the asset paths and directly reference the NVIDIA could location.感谢您使用我们的工作。仓库资产引用了“localhost”，该版本在下载 omnivores 启动器并在其中安装 nucleus 时存在于以前的版本中。这也应该让它为您服务。更好的解决方案是更新所有资产路径并直接引用 NVIDIA 可能的位置。

[image]
Although there has the asset, it doesnt work.
Could you please tell me where i can change the texture path? 
Thank you so much

pascal-roth, 2025-04-21:
You can convert the usd to usda and then change the path in the files or change each path individually in the Isaac GUI

pascal-roth, 2025-07-08:
@lrt20021018 I uploaded a new asset in PR #98, please see if that fixes your issues