leggedrobotics/viplanner #94: [Question] Question title

KingWang569, 2025-06-02:
Hey, friend
I encountered such a problem. When I downloaded the matterport3d dataset, I used python download_mp.py --task habitat -o /home/zzy/data/scene_datasets/mp3d to download it. After the download was completed, it was a 15g mp3d_habitat.zip file
After I unzipped this file, there were only the files in the figure below, and no matterport_mesh/xxx.obj file was found.
How should I download these files? Please help me.
Thank you very much!!!

[image]

[image]

ducmngx, 2025-06-05:
Any update on this issue?

pascal-roth, 2025-07-08:
Hi @ducmngx and @dongzhue, 

We did not use the Habitat Engine to download the files, but instead directly applied for the data in https://github.com/niessner/Matterport and then used the script sent to us when access to the data was granted.

Hope this clarifies the missing data links