leggedrobotics/viplanner #105: [Question]how to make the dog move

MaJiazhen8696, 2025-08-09:
[image]

I downloaded the model and config files as instructed. However, I couldn't find the path `${USER_PATH_TO_USD}` mentioned in viplanner/omniverse/README.md under the "Planer Demo" section to adjust in the corresponding config files.
I proceeded to run the demo with the following command as mentioned :
```
cd IsaacLab
./isaaclab.sh -p <path-to-viplanner-repo>/omniverse/standalone/viplanner_demo.py --scene warehouse --model_dir <path-to-model-download-dir>
```
The simulation window launched successfully, and the robot dog started walking for a few steps, but then it stopped and remained stationary. Interestingly, I could still see the predicted trajectory changing in the visualization, which suggests that the planner might still be working.

I noticed the following warning in the console:
`[Warning] [omni.isaac.lab.envs.manager_based_env] The render interval (1) is smaller than the decimation (20). Multiple multiple render calls will happen for each environment step. If this is not intended, set the render interval to be equal to the decimation.`
Any help would be greatly appreciated. Thank you!

brian00715, 2025-08-26:
You can drag the red cube using the mouse to change the target pose, then you can see the dog moving.

linden713, 2025-09-19:
Hi @MaJiazhen8696 
I encounter the same issue:

> The simulation window launched successfully, and the robot dog started walking for a few steps, but then it stopped and remained stationary. 

I try to change red cube position but the dog still not move.
Do you have any good solutions? Thanks in advance!

MaJiazhen8696, 2025-09-20:
> Hi [@MaJiazhen8696](https://github.com/MaJiazhen8696) I encounter the same issue:
>
>
> I try to change red cube position but the dog still not move. Do you have any good solutions? Thanks in advance!

@linden713 如果你打开后的场景是空白的(只有蓝色的网格和机器狗)，打开代理再试试，应该是需要从nvidia下载相关的资产，你也可以用omniverse nucelus创建本地服务器加载资产